ARMmbed/yotta #717: yotta init is not script friendly.

mjs-arm, 2016-02-24:
yotta init is hardwired to suck input from stdin and provides no options to allow override of that behaviour, this makes scripting around yotta init painful.  It would be useful to provide yotta init with a set of options, each corresponding to one of the prompts emitted to stdin and another option that indicates to accept the default values without prompting.

thegecko, 2016-02-24:
Something like an an answer file passed during init, perhaps?

As ```yotta init``` just creates the folder structure and module.json file, when scripting it we just create these folders and the json file as part on the script instead of calling ```yotta init```. Would this approach suffice in your use case?

mjs-arm, 2016-02-24:
Introducing an answer file is a new concept, simplicity of concept and principle of least surprise would both favour using options instead.

The concepts of Eencapsulation and information hiding, if applied, would have yotta do its own dirty work rather than requiring any script needing to achieve that same goal re-implement the behaviour.

bridadan, 2016-02-24:
While this does not solve your problem with command line config of the module, you can run yotta with `--noninteractive`. This will cause yotta to use all defaults.

I tried this as `yotta --noninteractive init` and it created the directories and module.json just fine. The default contents of some of the fields in the module.json are a little strange though:

```
{
  "name": "test-yt",
  "version": "0.0.0",
  "description": "A short description of what your module does goes here.",
  "keywords": [],
  "author": "Your Name <youremail@yourdomain.com>",
  "repository": {
    "url": "git@github.com:yourName/test-yt",
    "type": "git"
  },
  "homepage": "test-yt-module-homepage.com",
  "license": "Apache-2.0",
  "dependencies": {}
}
```

How would you like to add the config over the command line? Avoiding an option for each field would probably be best (there's already 17 possible fields).