ARMmbed/yotta #718: yotta command to get module dependencies directly from registry.

mjs-arm, 2016-02-24:
There appears to be no mechanism to get the dependencies of a module from the registry without using yotta to download and install a module along with all of its dependencies?

bogdanm, 2016-02-24:
That would be quite difficult, since a module can have target dependencies, so you'd have to select a target before getting the full list of dependencies.

mjs-arm, 2016-02-24:
.. or make the target a parameter of the query. Queries against the yotta registry do not need to be related to a local yotta install tree.

bogdanm, 2016-02-24:
That could be implemented, yes, but I'm not sure I understand the use case.

autopulated, 2016-02-25:
The web API supports this:

https://registry.yottabuild.org/modules/mbed-hal/versions/1.0.0

Or even:
https://registry.yottabuild.org/modules/mbed-hal/versions