ARMmbed/yotta #722: yotta version - error if already exists, but amends the commit

0xc0170, 2016-02-26:
I am running older yotta (0.12.2):

```
..\mbed-drivers (master)
λ yt version patch
info: @0.12.2
error: command failed: fatal: tag 'v0.12.2' already exists

C:\Code\git_repo\github\yotta_thing\mbed-drivers (master)
λ git logg -4
* f0ff915 - (HEAD, master) version v0.12.2 (68 seconds ago) <Martin Kojtal>
```

git reflog shows `` f0ff915 HEAD@{0}: commit: version v0.12.2 ``, which should not happen. As 0.12.2 was already on master, with a tag.

autopulated, 2016-02-26:
the tag is created after the commit (it has to be, because it's the commit that's tagged!), but we could check for the existence of the tag first.

In this case was the module.json already set to the correct version, presumably not? So there was also some sort of preceding error which meant the module.json was at 0.12.1, but there was an existing tag for 0.12.2?

0xc0170, 2016-03-02:
> the tag is created after the commit (it has to be, because it's the commit that's tagged!), but we could check for the existence of the tag first.

Exactly what I am proposing here. Otherwise a user needs to revert the last commit.

> In this case was the module.json already set to the correct version, presumably not? So there was also some sort of preceding error which meant the module.json was at 0.12.1, but there was an existing tag for 0.12.2?

This was a user error, the tag was pushed upstream, but not the commit.