ARMmbed/yotta #727: yotta gives error when redirecting output

bogdanm, 2016-03-04:
```
$ yt --plain search -s --limit 10000 mbed-hal-nordic > a
Fatal Exception, yotta=0.14.1
Traceback (most recent call last):
  File "/usr/local/bin/yt", line 4, in <module>
    yotta.main()
  File "/usr/local/lib/python2.7/dist-packages/yotta/main.py", line 201, in main
    status = args.command(args, following_args)
  File "/usr/local/lib/python2.7/dist-packages/yotta/search.py", line 128, in execCommand
    print(formatResult(result, args.plain, short=args.short))
UnicodeEncodeError: 'ascii' codec can't encode character u'\u2014' in position 157: ordinal not in range(128)
```

Without redirection, the command works.

Environment: Linux Mint 17.1

bridadan, 2016-03-04:
Is this at all related to #657 ?

PrzemekWirkus, 2016-03-08:
When I try:
```
$ yt --plain search -s --limit 10000 mbed-hal-nordic
```

I got thrilling:
```
Traceback (most recent call last):
  File "C:\Python27\Scripts\yt-script.py", line 9, in <module>
    load_entry_point('yotta', 'console_scripts', 'yt')()
  File "c:\work\przemek\yotta\yotta\main.py", line 201, in main
    status = args.command(args, following_args)
  File "c:\work\przemek\yotta\yotta\search.py", line 128, in execCommand
    print(formatResult(result, args.plain, short=args.short))
  File "C:\Python27\lib\site-packages\colorama\ansitowin32.py", line 36, in write
    self.__convertor.write(text)
  File "C:\Python27\lib\site-packages\colorama\ansitowin32.py", line 137, in write
    self.write_and_convert(text)
  File "C:\Python27\lib\site-packages\colorama\ansitowin32.py", line 165, in write_and_convert
    self.write_plain_text(text, cursor, len(text))
  File "C:\Python27\lib\site-packages\colorama\ansitowin32.py", line 170, in write_plain_text
    self.wrapped.write(text[start:end])
  File "C:\Python27\lib\encodings\cp437.py", line 12, in encode
    return codecs.charmap_encode(input,errors,encoding_map)
UnicodeEncodeError: 'charmap' codec can't encode character u'\u2014' in position 157: character maps to <undefined>
```

bogdanm, 2016-03-08:
> Is this at all related to #657 ?

I don't think so, they look like different issues.
@PrzemekWirkus, it might be that you're using a terminal that doesn't support Unicode encodings. One more reason to fix this issue.

autopulated, 2016-03-14:
Setting `PYTHONIOENCODING="utf-8"` (for example) should let you work around this.

bogdanm, 2016-03-14:
Yup, that does the trick just fine, thanks!

closed 2016-03-14 by bogdanm