ARMmbed/yotta #729: Can't install from dependency from github

wolfeidau, 2016-03-07:
Just trying to add mbedtls to a new project.

Newly minted project with no dependencies, I try and install https://github.com/ARMmbed/mbedtls and get the error as seen below.

Any idea how to get this working?

```
$ yotta install ARMmbed/mbedtls#master
info: mbedtls, ARMmbed/mbedtls#master
info: get versions for mbedtls
warning: Using head of "master" branch for "mbedtls", not a tagged version
info: download mbedtls@master from GitHub ARMmbed/mbedtls
Fatal Exception, yotta=0.14.1
Traceback (most recent call last):
  File "/usr/local/bin/yotta", line 4, in <module>
    yotta.main()
  File "/usr/local/lib/python2.7/site-packages/yotta/main.py", line 201, in main
    status = args.command(args, following_args)
  File "/usr/local/lib/python2.7/site-packages/yotta/install.py", line 60, in execCommand
    return installComponentAsDependency(args, c)
  File "/usr/local/lib/python2.7/site-packages/yotta/install.py", line 159, in installComponentAsDependency
    working_directory = modules_dir
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/access.py", line 381, in satisfyVersion
    name, version_required, working_directory, type=type, inherit_shrinkwrap = inherit_shrinkwrap
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/access.py", line 317, in satisfyVersionByInstalling
    name, version_required, install_into, v, type=type, inherit_shrinkwrap = inherit_shrinkwrap
  File "/usr/local/lib/python2.7/site-packages/yotta/lib/access.py", line 332, in _satisfyVersionByInstallingVersion
    'Dependency "%s":"%s" is not a valid %s.' % (name, version_required, type)
Exception: Dependency "mbedtls":"ARMmbed/mbedtls#master" is not a valid module.
```

BlackstoneEngineering, 2016-03-07:
@wolfeidau Please give us the output of a `yt ls` and a `yt target` in your directory so we can try to reproduce.

bridadan, 2016-03-07:
The reason you can't install mbedtls straight from GitHub is because they generate multiple packages from that repository.

The recommended way to install mbedtls is from the yotta repository:

```
yotta install mbedtls
```

But if you need to use the latest code from the repo, you can generate the yotta module locally and link it to your application. See this section of the mbedtls README: https://github.com/ARMmbed/mbedtls#yotta

wolfeidau, 2016-03-08:
Not a problem, I just forked the code off and made my own repo with the contents of mbedtls at https://github.com/wolfeidau/mbedtls for testing, not ideal but it works.

Need this to work with a CI server, linking would be a disaster.

@bridadan The issue was I couldn't use mbedtls in the repository as it has a bug with certificate chains. As it is now released I can just use the package once i finish testing.

closed 2016-03-08 by wolfeidau