ARMmbed/yotta #731: error: this module is private and cannot be published

cyliangtw, 2016-03-09:
I could publish target module into yotta registry, like as nu-nuc472-nubed-armcc or sal-driver-lwip-nuvoton modules.
However, I can’t publish mbed-hal-nuc472 module with error message _“**error: this module is private and cannot be published**”_.
Even I changed this module as public this afternoon but still encountered the same error message.
Please guide me how to resolve it.

**mbed-hal-nuc472 module at:**
https://github.com/ARMmbed/mbed-hal-nuc472

_**Log as below:**_
```
$ yt owner add cyliang@nuvoton.com
error: no such module, "mbed-hal-nuc472"
$ yt publish
warning: You're publishing with the mbed-official tag. Is this really an officially supported mbed module? If not, please remove the mbed-official tag from your module.json file. If you are unsure, please ask on the issue tracker.
If you still want to publish, press [enter] to continue.
error: this module is private and cannot be published
```

bogdanm, 2016-03-09:
Hi,

Remove the `"private": true,` line from your `module.json` (https://github.com/ARMmbed/mbed-hal-nuc472/blob/master/module.json#L5), you should be able to publish after this.

cyliangtw, 2016-03-10:
Thanks, it works.

closed 2016-03-10 by cyliangtw