ARMmbed/yotta #744: SSL: CERTIFICATE_VERIFY_FAILED - OpenSSL Versioning and Certificate Locations

bitbangr, 2016-03-19:
Running OS X 10.10.5 Yosemite running on Macbook Air
used installer yotta_osx_installer_v0.0.4.dmg

Got the following error when running  -> yotta search target "x86-osx-native" 

`(yotta workspace) ~/Documents/VHS/oshchip/blinky $ yotta search target "x86-osx-native"
Traceback (most recent call last):
  File "/Applications/yotta.app/Contents/Resources/workspace/bin/yotta", line 4, in <module>
    yotta.main()
  File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/yotta/main.py", line 239, in main
    status = args.command(args, following_args)
  File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/yotta/search.py", line 53, in execCommand
    for result in registry_access.search(query=args.query, keywords=args.kw, registry=args.registry):
  File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/yotta/lib/registry_access.py", line 679, in search
    response = requests.get(url, headers=headers, params=params)
  File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/requests/api.py", line 69, in get
    return request('get', url, params=params, **kwargs)
  File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/requests/api.py", line 50, in request
    response = session.request(method=method, url=url, **kwargs)
  File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/requests/sessions.py", line 471, in request
    resp = self.send(prep, **send_kwargs)
  File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/requests/sessions.py", line 579, in send
    r = adapter.send(request, **kwargs)
  File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/requests/adapters.py", line 430, in send
    raise SSLError(e, request=request)
requests.exceptions.SSLError: ("bad handshake: Error([('SSL routines', 'ssl3_get_server_certificate', 'certificate verify failed')],)",)
`

The Yotta workspace Python is using OpenSSL 0.9.8zg 14 July 2015 as per below check

`(yotta workspace) ~/Documents/VHS/oshchip/blinky $ python -c 'import ssl; print(ssl.OPENSSL_VERSION)'
`
OpenSSL 0.9.8zg 14 July 2015

---------------------

brew info openssl shows -> openssl: stable 1.0.2e (bottled) [keg-only]

This is a different version that what the yotta workspace is using. 
The Brew info Caveats were also interesting to note.

`(yotta workspace) ~/Documents/VHS/oshchip/blinky $ brew info openssl
openssl: stable 1.0.2e (bottled) [keg-only]
SSL/TLS cryptography library
https://openssl.org/
/usr/local/Cellar/openssl/1.0.2d_1 (464 files, 12M)
  Poured from bottle
From: https://github.com/Homebrew/homebrew/blob/master/Library/Formula/openssl.rb
==> Dependencies
Build: makedepend ✘
==> Options
--universal
	Build a universal binary
--without-test
	Skip build-time tests (not recommended)
==> Caveats
A CA file has been bootstrapped using certificates from the system
keychain. To add additional certificates, place .pem files in
  /usr/local/etc/openssl/certs

and run
  /usr/local/opt/openssl/bin/c_rehash

This formula is keg-only, which means it was not symlinked into /usr/local.

Apple has deprecated use of OpenSSL in favor of its own TLS and crypto libraries`

--------------------------
So Two Issues:

1) The yotta install is apparently not using the latest version of openssl
2) The Yotta supplied pem certs may possibly not be accessible as per above Caveats.  

I searched for pem files and found the following
`root@myairride~$ find / -name 'cacert.pem'

/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/pip/_vendor/requests/cacert.pem
/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/requests/cacert.pem`

This issue also appears to be the related to issue [SSL: CERTIFICATE_VERIFY_FAILED #391](https://github.com/ARMmbed/yotta/issues/391)

I would do a brew update openssl but not sure how this would work or affect the Yotta Workspace VM.

Mark

bitbangr, 2016-03-19:
There seems to be an issue with the Certificates for the domains that require network access
http://yottadocs.mbed.com/reference/registry.html#network-access

https://registry.yottabuild.org

(yotta workspace) ~/Documents/VHS/oshchip/blinky $ wget https://registry.yottabuild.org
--2016-03-18 17:51:28--  https://registry.yottabuild.org/
Resolving registry.yottabuild.org... 54.243.54.152, 54.235.218.129, 107.20.190.36
Connecting to registry.yottabuild.org|54.243.54.152|:443... connected.
ERROR: cannot verify registry.yottabuild.org's certificate, issued by 'CN=COMODO RSA Domain Validation Secure Server CA,O=COMODO CA Limited,L=Salford,ST=Greater Manchester,C=GB':
  Unable to locally verify the issuer's authority.
To connect to registry.yottabuild.org insecurely, use `--no-check-certificate'.

autopulated, 2016-03-21:
The registry has a new cert (since last week), but it's definitely valid and installed correctly. Possibly you have a very old set of root certificates which is being used.

Note that the yottabuild.org root domain redirects, so to test the certificate in your browser you need to load https://registry.yottabuild.org.

Please could you try using the latest master version of yotta, which includes an explicit dependency on the [certifi](https://pypi.python.org/pypi/certifi) set of root certificates. (Alternatively, running `pip install -U certifi` would test the same thing).

bitbangr, 2016-03-21:
I've installed and used the latest master version of yotta and I am now able to do yotta builds, targets etc. with no SSL certificate errors!

You can close this issue if you like.

I am guessing that the downloadable **OS X yotta.app** in the **# Using yotta.app** section of the [http://yottadocs.mbed.com/#installing](http://yottadocs.mbed.com/#installing) web page needs to be updated.  

Possibly the following certs contained in the OS X Yotta App need updating? 
/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/pip/_vendor/requests/cacert.pem
/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/requests/cacert.pem`

closed 2016-03-21 by bitbangr

anderss90, 2016-08-18:
autopulated's answer solved this for me. ( i ran pip install -U certifi).