ARMmbed/yotta #758: v0.16.0 version installation issue

bcostm, 2016-03-31:
I’ve tried to install yotta v0.16.0 and I got this error:

`Installed c:\python27\lib\site-packages\yotta-0.16.0-py2.7.egg
Processing dependencies for yotta==0.16.0
Traceback (most recent call last):
  File "setup.py", line 71, in <module>
    ] + platform_deps
  File "C:\Python27\lib\distutils\core.py", line 151, in setup
    dist.run_commands()
  File "C:\Python27\lib\distutils\dist.py", line 953, in run_commands
    self.run_command(cmd)
  File "C:\Python27\lib\distutils\dist.py", line 972, in run_command
    cmd_obj.run()
  File "C:\Python27\lib\site-packages\setuptools\command\install.py", line 67, in run
    self.do_egg_install()
  File "C:\Python27\lib\site-packages\setuptools\command\install.py", line 117, in do_egg_install
    cmd.run()
  File "C:\Python27\lib\site-packages\setuptools\command\easy_install.py", line 391, in run
    self.easy_install(spec, not self.no_deps)
  File "C:\Python27\lib\site-packages\setuptools\command\easy_install.py", line 621, in easy_install
    return self.install_item(None, spec, tmpdir, deps, True)
  File "C:\Python27\lib\site-packages\setuptools\command\easy_install.py", line 672, in install_item
    self.process_distribution(spec, dist, deps)
  File "C:\Python27\lib\site-packages\setuptools\command\easy_install.py", line 717, in process_distribution
    [requirement], self.local_index, self.easy_install
  File "C:\Python27\lib\site-packages\pkg_resources\__init__.py", line 814, in resolve
    if req.marker and not req.marker.evaluate():
  File "C:\Python27\lib\site-packages\pkg_resources\_vendor\packaging\markers.py", line 278, in evaluate
    return _evaluate_markers(self._markers, current_environment)
  File "C:\Python27\lib\site-packages\pkg_resources\_vendor\packaging\markers.py", line 203, in _evaluate_markers
    lhs_value = _get_env(environment, lhs.value)
  File "C:\Python27\lib\site-packages\pkg_resources\_vendor\packaging\markers.py", line 185, in _get_env
    "{0!r} does not exist in evaluation environment.".format(name)
pkg_resources._vendor.packaging.markers.UndefinedEnvironmentName: 'extra' does not exist in evaluation environment.`

Can you help ? Thanks.

autopulated, 2016-03-31:
it looks like you're using `easy_install`. Try `easy_install pip`, then using pip to install yotta?

bcostm, 2016-03-31:
I tried with pip and I have the same errors...

bcostm, 2016-03-31:
I discovered that the error does not come with the installation but when I type `yotta --version` or `yotta --help`. Do you have the same thing ?

autopulated, 2016-03-31:
OK thanks for the clarification. I can reproduce the problem, and I think #760 should work around it, but the root cause remains a mystery: for some reason when pkg_resources is resolving yotta's dependencies it's choking on pyasn1 (a dependency of requests[security]). I can reproduce this without yotta by running:

```
python -c "import pkg_resources; pkg_resources.require('requests[security]')"
```

However nothing related to that dependency has changed recently, so I don't know why this problem was first reported today.

autopulated, 2016-03-31:
(actually, looks like it was setuptools 20.6.6 which broke things, `pip install setuptools==20.4` will revert to the previous published version, which works)

autopulated, 2016-03-31:
Fixed by https://github.com/ARMmbed/yotta/releases/tag/v0.16.1

closed 2016-03-31 by autopulated

bcostm, 2016-03-31:
Thanks for the patch for I still have the same problem after installing the v0.16.1.

But I tried `pip install setuptools==20.4` after the v0.16.1 was installed and now it is ok.

autopulated, 2016-03-31:
OK thanks for reporting back.

autopulated, 2016-03-31:
setuptools v20.6.7 has now also been released, which reverts the underlying bug.