ARMmbed/yotta #759: yotta update fails after upgrading to yotta 0.16.0

kmdewaal, 2016-03-31:
After upgrading to yotta 0.16.0 the "yotta update" command fails.
The "yotta version" and "yotta build" commands execute correct.

```
(workspace) C:\_ddm\openais-eu-project\openais-mbed-philips>yotta --version
0.16.0

(workspace) C:\_ddm\openais-eu-project\openais-mbed-philips>yotta update
info: get versions for frdm-k64f-gcc
Fatal Exception, yotta=unknown
Traceback (most recent call last):
  File "c:\yotta\python\Lib\runpy.py", line 162, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "c:\yotta\python\Lib\runpy.py", line 72, in _run_code
    exec code in run_globals
  File "C:\yotta\workspace\Scripts\yotta.exe\__main__.py", line 9, in <module>
  File "c:\yotta\workspace\lib\site-packages\yotta\main.py", line 44, in wrapped
    return fn(*args, **kwargs)
  File "c:\yotta\workspace\lib\site-packages\yotta\main.py", line 225, in main
    status = args.command(args, following_args)
  File "c:\yotta\workspace\lib\site-packages\yotta\update.py", line 37, in execCommand
    target, errors = c.satisfyTarget(args.target, update_installed=True, additional_config=args.config)
  File "c:\yotta\workspace\lib\site-packages\yotta\lib\component.py", line 677, in satisfyTarget
    shrinkwrap = self.getShrinkwrap()
  File "c:\yotta\workspace\lib\site-packages\yotta\lib\target.py", line 156, in getDerivedTarget
    inherit_shrinkwrap = shrinkwrap
  File "c:\yotta\workspace\lib\site-packages\yotta\lib\access.py", line 379, in satisfyVersion
    inherit_shrinkwrap = inherit_shrinkwrap
  File "c:\yotta\workspace\lib\site-packages\yotta\lib\access.py", line 282, in satisfyVersionFromSearchPaths
    v = latestSuitableVersion(name, version_required, registry=_registryNamespaceForType(type))
  File "c:\yotta\workspace\lib\site-packages\yotta\lib\access.py", line 107, in latestSuitableVersion
    vers = remote_component.availableVersions()
  File "c:\yotta\workspace\lib\site-packages\yotta\lib\registry_access.py", line 485, in availableVersions
    return _listVersions(self.namespace, self.name)
  File "c:\yotta\workspace\lib\site-packages\yotta\lib\registry_access.py", line 100, in wrapped
    return fn(*args, **kwargs)
  File "c:\yotta\workspace\lib\site-packages\yotta\lib\registry_access.py", line 232, in _listVersions
    request_headers = _headersForRegistry(registry)
  File "c:\yotta\workspace\lib\site-packages\yotta\lib\registry_access.py", line 414, in _headersForRegistry
    'X-Yotta-Client-Version': _getYottaVersion(),
  File "c:\yotta\workspace\lib\site-packages\yotta\lib\registry_access.py", line 397, in _getYottaVersion
    _yotta_version = pkg_resources.require("yotta")[0].version
  File "c:\yotta\workspace\lib\site-packages\pkg_resources\__init__.py", line 947, in require
    needed = self.resolve(parse_requirements(requirements))
  File "c:\yotta\workspace\lib\site-packages\pkg_resources\__init__.py", line 814, in resolve
    if req.marker and not req.marker.evaluate():
  File "c:\yotta\workspace\lib\site-packages\pkg_resources\_vendor\packaging\markers.py", line 278, in evaluate
    return _evaluate_markers(self._markers, current_environment)
  File "c:\yotta\workspace\lib\site-packages\pkg_resources\_vendor\packaging\markers.py", line 203, in _evaluate_markers
    lhs_value = _get_env(environment, lhs.value)
  File "c:\yotta\workspace\lib\site-packages\pkg_resources\_vendor\packaging\markers.py", line 185, in _get_env
    "{0!r} does not exist in evaluation environment.".format(name)
pkg_resources._vendor.packaging.markers.UndefinedEnvironmentName: 'extra' does not exist in evaluation environment.

(workspace) C:\_ddm\openais-eu-project\openais-mbed-philips>

```

autopulated, 2016-03-31:
Run  `pip install setuptools==20.4` to workaround this temporarily. A version of yotta which works around this will be out soon.

autopulated, 2016-03-31:
Fixed by https://github.com/ARMmbed/yotta/releases/tag/v0.16.1

closed 2016-03-31 by autopulated