ARMmbed/yotta #761: "yotta debug" error

ccli8, 2016-04-01:
Hi,

yotta: 0.16.1

Error message
```
info: found hello-numbed at source\hello-numbed
Invalid parameter(s)
RESET { SESSION }
Fatal Exception, yotta=0.16.1
Traceback (most recent call last):
  File "C:\Python27\Scripts\yotta-script.py", line 9, in <module>
    load_entry_point('yotta==0.16.1', 'console_scripts', 'yotta')()
  File "c:\python27\lib\site-packages\yotta\main.py", line 44, in wrapped
    return fn(*args, **kwargs)
  File "c:\python27\lib\site-packages\yotta\main.py", line 225, in main
    status = args.command(args, following_args)
  File "c:\python27\lib\site-packages\yotta\debug.py", line 51, in execCommand
    error = target.debug(builddir, args.program)
  File "c:\python27\lib\site-packages\yotta\lib\target.py", line 634, in debug
    return self._debugWithScript(builddir, program)
  File "c:\python27\lib\site-packages\yotta\lib\fsutils_win.py", line 23, in wrapper
    return fn(*args, **kwargs)
  File "c:\python27\lib\site-packages\yotta\lib\target.py", line 665, in _debugWithScript
    cmd, cwd = builddir, env = env
  File "c:\python27\lib\subprocess.py", line 710, in __init__
    errread, errwrite)
  File "c:\python27\lib\subprocess.py", line 958, in _execute_child
    startupinfo)
TypeError: environment can only contain strings
```

If I convert _prog_path_ in lib/target.py as below, the error get fixed.
```
    def _debugWithScript(self, builddir, program):
        child = None
        try:
            prog_path = self.findProgram(builddir, program)
            if prog_path is None:
                return

            env = os.environ.copy()
            #env['YOTTA_PROGRAM'] = prog_path
            env['YOTTA_PROGRAM'] = str(prog_path)
```

autopulated, 2016-04-01:
Thanks – which python version are you using?

ccli8, 2016-04-01:
python: 2.7.11

closed 2016-04-01 by autopulated

autopulated, 2016-04-01:
yotta 0.16.2, just released, should fix this

ccli8, 2016-04-06:
Yes. yotta 0.16.2 fixed the issue. Thanks.

PR #762 (2016-04-01): ensure unicode paths don't end up subprocess environments under python 2
Fixes #761