ARMmbed/yotta #763: No way to authenticate automatic builds

drewcassidy, 2016-04-02:
With the current system, doing anything (including downloading new modules) in yotta requires the user to manually log in for that machine. This makes continuous integration with services like Travis CI impossible to use with yotta. perhaps allow for the user to authorize themselves with an OAuth token from either Github or mbed?

autopulated, 2016-04-02:
It isn't actually the oauth token that matters in this case, it's that the public/private key pair in your yotta settings (~/.yotta/config.json) has been associated with an email address via a login flow.

If you log in once with a public/private key pair, then put that in the settings file for your automated builds, things will work without any further authentication.

(Note: never share a private key publicly that is associated with an email address you use to publish packages: always use one associated with a dummy email address you use for nothing else)

Hope this is somewhat helpful, really sorry about the inconvenience caused by the need to log in to download things: it is quite contrary to the principle in yotta that it should be easy to test things :(

larsdunemarkhms, 2016-04-07:
@drewcassidy one solution that I found was to add YOTTA_KEYS_PUBLIC and YOTTA_KEYS_PRIVATE to the build job environment variables.
Then you don't need to bundle your public and private keys to a build node image. Instead it can be injected at build time.

autopulated, 2016-04-07:
Yup that should work: everything in yotta's config can also be set by environment variables (names converted to uppercase, prefixed by `YOTTA_`, and fields separated by `_`)

drewcassidy, 2016-04-07:
Thanks. this solves building and deploying in the main repository. 

Unfortunately, the Travis secure environment variables are unavailable to forks, since someone could modify the build script to steal them, then make a pull request. I obviously dont want to deploy from pull requests, but I would like to have pull requests get tested in travis, which is impossible since I need to be able to install targets and mbed-drivers. 

Is there really no way to install modules/targets through yotta without authorization? I could pull them directly from github since they are all open source, but I would like to keep this as simple as possible

autopulated, 2016-04-07:
yotta's own tests use this dummy config file: https://github.com/ARMmbed/yotta/blob/master/.yotta_test_config.json
which you could also use (they keys and token there are for an account used for nothing else).

drewcassidy, 2016-04-08:
Thanks for that. I'll give it a try for non-publish builds. Feel free to close this issue, but I still believe auth shouldn't be necessary for installing modules.

By the way, maybe try contacting Travis to see if you can get Yotta added as a built in deployment provider

closed 2017-03-28 by thegecko