ARMmbed/yotta #764: yt target x86-linux-native fails with 'EntryPoint' object has no attribute 'resolve'

janjongboom, 2016-04-07:
On a fresh machine, just installed yotta...

```
janjongboom:~/workspace/mbed-client-for-lora (master) $ yt target x86-linux-native
info: get versions for x86-linux-native
Fatal Exception, yotta=0.16.2
Traceback (most recent call last):
  File "/usr/local/bin/yt", line 4, in <module>
    yotta.main()
  File "/usr/local/lib/python2.7/dist-packages/yotta/main.py", line 44, in wrapped
    return fn(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/yotta/main.py", line 225, in main
    status = args.command(args, following_args)
  File "/usr/local/lib/python2.7/dist-packages/yotta/target.py", line 126, in execCommand
    target, errors = c.satisfyTarget(t)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/component.py", line 677, in satisfyTarget
    shrinkwrap = self.getShrinkwrap()
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/target.py", line 166, in getDerivedTarget
    inherit_shrinkwrap = shrinkwrap
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/access.py", line 385, in satisfyVersion
    name, version_required, working_directory, type=type, inherit_shrinkwrap = inherit_shrinkwrap
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/access.py", line 314, in satisfyVersionByInstalling
    v = latestSuitableVersion(name, version_required, _registryNamespaceForType(type))
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/access.py", line 107, in latestSuitableVersion
    vers = remote_component.availableVersions()
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/registry_access.py", line 481, in availableVersions
    return _listVersions(self.namespace, self.name)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/registry_access.py", line 100, in wrapped
    return fn(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/registry_access.py", line 232, in _listVersions
    request_headers = _headersForRegistry(registry)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/registry_access.py", line 406, in _headersForRegistry
    auth_token = generate_jwt_token(_getPrivateKeyObject(registry), registry)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/registry_access.py", line 375, in _getPrivateKeyObject
    pubkey_pem, privatekey_pem = _generateAndSaveKeys(registry)
  File "/usr/local/lib/python2.7/dist-packages/yotta/lib/registry_access.py", line 334, in _generateAndSaveKeys
    public_exponent=65537, key_size=2048, backend=default_backend()
  File "/usr/local/lib/python2.7/dist-packages/cryptography/hazmat/backends/__init__.py", line 35, in default_backend
    _default_backend = MultiBackend(_available_backends())
  File "/usr/local/lib/python2.7/dist-packages/cryptography/hazmat/backends/__init__.py", line 22, in _available_backends
    "cryptography.backends"
AttributeError: 'EntryPoint' object has no attribute 'resolve'
```

autopulated, 2016-04-07:
Looks like it might be related to this: https://github.com/pyca/cryptography/issues/2853

What does `pip list | grep setuptools` say?

janjongboom, 2016-04-07:
I removed the VM... I can check tomorrow.

janjongboom, 2016-04-08:
@autopulated janjongboom1:~/workspace $ pip list | grep setuptools
setuptools (3.3)

yogpan01, 2016-04-11:
@autopulated I am also having the same issue as @janjongboom , my setuptools version is also 3.3

Eilliar, 2016-04-13:
Same Problem here..... setuptools (3.3).

:(

yogpan01, 2016-04-13:
Downgrading yotta to 0.16.0 works for me.

autopulated, 2016-04-13:
It isn't clear why downgrading to 0.16.0 would solve this, as none of the versions that yotta requires of its dependencies have changed between 0.16.0 and 0.16.2: if you try updating to 0.16.2 does it still work?

In any case the root cause appears to be old versions of setuptools: cryptography already has a dependency on [>=11.3](https://github.com/pyca/cryptography/blob/master/setup.py#L40), which for some reason hasn't been satisfied in all the cases we're seeing this problem.

Eilliar, 2016-04-13:
Even Downgrading it to 0.16.0 does not work....

autopulated, 2016-04-13:
`pip install -U setuptools` should solve things, provided you can do that successfully.

voonter, 2016-04-16:
Installing newest version of setuptools helps me. I think that it could be done by: 

`sudo -H pip install yotta--upgrade`

swkim01, 2016-04-17:
It seems to be conflicting package module name. You have to remove pkg_resources.pyc and rename pkg_resources.py to pkg_resources.py.old in "/usr/lib/python2.7/dist-packages".

ChinaShrimp, 2016-04-29:
Hi, is there any update regarding this issue, none solutions mentioned above work for me...

LiyouZhou, 2016-06-15:
Was trying to install yotta on c9. following the normal steps provided in yotta docs, found same error.

autopulated, 2016-06-15:
AFAIK this is only caused by old versions of setuptools, so double check that you have the latest versions!

See [here for a handy explanation](http://stackoverflow.com/a/14753678/233201) of the different modules you might see (distribute vs. setuptools) with an old base installation that you're trying to update.

janjongboom, 2016-06-27:
Ran into this today again on Raspberry Pi Zero. Today `sudo easy_install setuptools` did the trick.

peisenhower, 2016-08-11:
having same issue. None of the resolutions here worked for me. Also running on WSL (windows subsystem for linux). Which is probably adding a ton of trouble to this.

thegecko, 2017-03-28:
Closing this issue as there is a workaround

closed 2017-03-28 by thegecko