ARMmbed/yotta #766: yotta (0.15.2) build / target throws fatal error on Fedora 22

shahab5, 2016-04-14:
yotta build

info: get versions for x86-linux-native
Fatal Exception, yotta=unknown
Traceback (most recent call last):
  File "/usr/bin/yotta", line 4, in <module>
    yotta.main()
  File "/usr/lib/python2.7/site-packages/yotta/main.py", line 44, in wrapped
    return fn(*args, **kwargs)
  File "/usr/lib/python2.7/site-packages/yotta/main.py", line 221, in main
    status = args.command(args, following_args)
  File "/usr/lib/python2.7/site-packages/yotta/build.py", line 46, in execCommand
    status = installAndBuild(args, following_args)
  File "/usr/lib/python2.7/site-packages/yotta/build.py", line 70, in installAndBuild
    target, errors = c.satisfyTarget(args.target, additional_config=args.config)
  File "/usr/lib/python2.7/site-packages/yotta/lib/component.py", line 677, in satisfyTarget
    shrinkwrap = self.getShrinkwrap()
  File "/usr/lib/python2.7/site-packages/yotta/lib/target.py", line 156, in getDerivedTarget
    inherit_shrinkwrap = shrinkwrap
  File "/usr/lib/python2.7/site-packages/yotta/lib/access.py", line 385, in satisfyVersion
    name, version_required, working_directory, type=type, inherit_shrinkwrap = inherit_shrinkwrap
  File "/usr/lib/python2.7/site-packages/yotta/lib/access.py", line 314, in satisfyVersionByInstalling
    v = latestSuitableVersion(name, version_required, _registryNamespaceForType(type))
  File "/usr/lib/python2.7/site-packages/yotta/lib/access.py", line 107, in latestSuitableVersion
    vers = remote_component.availableVersions()
  File "/usr/lib/python2.7/site-packages/yotta/lib/registry_access.py", line 485, in availableVersions
    return _listVersions(self.namespace, self.name)
  File "/usr/lib/python2.7/site-packages/yotta/lib/registry_access.py", line 100, in wrapped
    return fn(*args, **kwargs)
  File "/usr/lib/python2.7/site-packages/yotta/lib/registry_access.py", line 232, in _listVersions
    request_headers = _headersForRegistry(registry)
  File "/usr/lib/python2.7/site-packages/yotta/lib/registry_access.py", line 414, in _headersForRegistry
    'X-Yotta-Client-Version': _getYottaVersion(),
  File "/usr/lib/python2.7/site-packages/yotta/lib/registry_access.py", line 397, in _getYottaVersion
    _yotta_version = pkg_resources.require("yotta")[0].version
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 918, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 813, in resolve
    new_requirements = dist.requires(req.extras)[::-1]
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2558, in requires
    "%s has no such extra feature %r" % (self, ext)
pkg_resources.UnknownExtra: requests 2.6.2 has no such extra feature 'security'

shahab5, 2016-04-14:
I have checked,

1. system time got using python is in sync with UTC.
2. pip installation  "sudo pip install pyasn1 --upgrade" reports already up-to-date.

Thanks

autopulated, 2016-04-14:
Your version of the `requests` package is too old. Run `pip install -U requests` first.

shahab5, 2016-04-14:
@autopulated  Thanks for reply.Pip says its upto date.

pip install -U requests
Requirement already up-to-date: requests in /usr/lib/python2.7/site-packages

autopulated, 2016-04-14:
ah I misread the version in your error message – you do have >=2.4.1, but still only version 2.6.2 (the latest is 2.9.1).

So possibly this is another setuptools issue – It's worth trying installing the latest version of that (`pip install -U setuptools`), and also updating to the latest yotta, which doesn't use pgk_resources any more when running commands (`pip install -U yotta`).

shahab5, 2016-04-14:
@autopulated Updated pip using 

`dnf install python3-pip`

`pip install -U setuptools`

Collecting setuptools
  Using cached setuptools-20.7.0-py2.py3-none-any.whl
Installing collected packages: setuptools
  Found existing installation: setuptools 12.0.3
    Uninstalling setuptools-12.0.3:
      Successfully uninstalled setuptools-12.0.3
Successfully installed setuptools-20.7.0

Got a different Error installing yotta,
`pip install -U yotta`

Collecting yotta
  Downloading yotta-0.16.3.tar.gz (128kB)
    100% |████████████████████████████████| 133kB 202kB/s 
Requirement already up-to-date: semantic-version<3,>=2.3.1 in /usr/lib/python3.4/site-packages (from yotta)
Requirement already up-to-date: requests[security]<3,>=2.4.3 in /usr/lib/python3.4/site-packages (from yotta)
Requirement already up-to-date: certifi>=2016.2.28 in /usr/lib/python3.4/site-packages (from yotta)
Collecting PyGithub<2,>=1.25 (from yotta)
  Using cached PyGithub-1.26.0.tar.gz
Requirement already up-to-date: colorama<0.4,>=0.3 in /usr/lib/python3.4/site-packages (from yotta)
Requirement already up-to-date: hgapi<2,>=1.7 in /usr/lib/python3.4/site-packages (from yotta)
Collecting Jinja2<3,>=2.7.0 (from yotta)
  Using cached Jinja2-2.8-py2.py3-none-any.whl
Collecting cryptography>=0.8 (from yotta)
  Using cached cryptography-1.3.1.tar.gz
Collecting PyJWT<2.0,>=1.0 (from yotta)
  Using cached PyJWT-1.4.0-py2.py3-none-any.whl
Collecting pathlib<1.1,>=1.0.1 (from yotta)
  Using cached pathlib-1.0.1.tar.gz
Collecting jsonschema<3.0,>=2.4.0 (from yotta)
  Using cached jsonschema-2.5.1-py2.py3-none-any.whl
Collecting argcomplete<2.0,>=0.8.0 (from yotta)
  Using cached argcomplete-1.1.1-py2.py3-none-any.whl
Collecting mbed-test-wrapper<2.0.0,>=0.0.3 (from yotta)
  Using cached mbed_test_wrapper-1.0.0.tar.gz
Collecting valinor<1.0,>=0.0.0 (from yotta)
  Using cached valinor-0.0.15.tar.gz
Collecting intelhex<3.0,>=2.0 (from yotta)
  Using cached intelhex-2.1-py2.py3-none-any.whl
Collecting jsonpointer<2.0,>=1.9 (from yotta)
  Using cached jsonpointer-1.10.tar.gz
Requirement already up-to-date: pyasn1 in /usr/lib/python3.4/site-packages (from requests[security]<3,>=2.4.3->yotta)
Collecting ndg-httpsclient (from requests[security]<3,>=2.4.3->yotta)
  Using cached ndg_httpsclient-0.4.0.tar.gz
Collecting pyOpenSSL>=0.13 (from requests[security]<3,>=2.4.3->yotta)
  Using cached pyOpenSSL-16.0.0-py2.py3-none-any.whl
Requirement already up-to-date: MarkupSafe in /usr/lib/python3.4/site-packages (from Jinja2<3,>=2.7.0->yotta)
Requirement already up-to-date: idna>=2.0 in /usr/lib/python3.4/site-packages (from cryptography>=0.8->yotta)
Collecting six>=1.4.1 (from cryptography>=0.8->yotta)
  Using cached six-1.10.0-py2.py3-none-any.whl
Requirement already up-to-date: setuptools>=11.3 in /usr/lib/python3.4/site-packages (from cryptography>=0.8->yotta)
Collecting cffi>=1.4.1 (from cryptography>=0.8->yotta)
  Using cached cffi-1.5.2.tar.gz
Requirement already up-to-date: pyyaml<4,>=3 in /usr/lib/python3.4/site-packages (from valinor<1.0,>=0.0.0->yotta)
Collecting pyOCD<1.0,>=0.3 (from valinor<1.0,>=0.0.0->yotta)
  Using cached pyOCD-0.7.0.zip
Collecting project-generator<0.9.0,>=0.8.0 (from valinor<1.0,>=0.0.0->yotta)
  Using cached project_generator-0.8.14.zip
Collecting pyelftools==0.23 (from valinor<1.0,>=0.0.0->yotta)
  Using cached pyelftools-0.23.tar.gz
Requirement already up-to-date: pycparser in /usr/lib/python3.4/site-packages (from cffi>=1.4.1->cryptography>=0.8->yotta)
Collecting enum34 (from pyOCD<1.0,>=0.3->valinor<1.0,>=0.0.0->yotta)
  Using cached enum34-1.1.3-py2.py3-none-any.whl
Collecting pyusb>=1.0.0b2 (from pyOCD<1.0,>=0.3->valinor<1.0,>=0.0.0->yotta)
[... 30 lines cut ...]
    copying cffi/__init__.py -> build/lib.linux-i686-3.4/cffi
    copying cffi/lock.py -> build/lib.linux-i686-3.4/cffi
    copying cffi/api.py -> build/lib.linux-i686-3.4/cffi
    copying cffi/gc_weakref.py -> build/lib.linux-i686-3.4/cffi
    copying cffi/vengine_gen.py -> build/lib.linux-i686-3.4/cffi
    copying cffi/ffiplatform.py -> build/lib.linux-i686-3.4/cffi
    copying cffi/cparser.py -> build/lib.linux-i686-3.4/cffi
    copying cffi/backend_ctypes.py -> build/lib.linux-i686-3.4/cffi
    copying cffi/commontypes.py -> build/lib.linux-i686-3.4/cffi
    copying cffi/model.py -> build/lib.linux-i686-3.4/cffi
    copying cffi/vengine_cpy.py -> build/lib.linux-i686-3.4/cffi
    copying cffi/_cffi_include.h -> build/lib.linux-i686-3.4/cffi
    copying cffi/parse_c_type.h -> build/lib.linux-i686-3.4/cffi
    copying cffi/_embedding.h -> build/lib.linux-i686-3.4/cffi
    running build_ext
    building '_cffi_backend' extension
    creating build/temp.linux-i686-3.4
    creating build/temp.linux-i686-3.4/c
    gcc -pthread -Wno-unused-result -DDYNAMIC_ANNOTATIONS_ENABLED=1 -DNDEBUG -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -m32 -march=i686 -mtune=atom -fasynchronous-unwind-tables -D_GNU_SOURCE -fPIC -fwrapv -fPIC -DUSE__THREAD -I/usr/include/python3.4m -c c/_cffi_backend.c -o build/temp.linux-i686-3.4/c/_cffi_backend.o
    c/_cffi_backend.c:2:20: fatal error: Python.h: No such file or directory
    compilation terminated.
    error: command 'gcc' failed with exit status 1
    
    ----------------------------------------
**Command "/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-la32wpnl/cffi/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-ce2zv5c3-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-la32wpnl/cffi/**

shahab5, 2016-04-14:
How to remove the existing yotta (0.15.2), 

`pip uninstall yotta`
Cannot uninstall requirement yotta, not installed

But yotta command exists.

autopulated, 2016-04-14:
`which yotta` will tell you where the yotta that's being run is:

It's possible that you have two separate python installations – python 2, and python 3. I'm not sure how Fedora handles this, but you might have separate pip2 and pip3 commands for installing/uninstalling programs for each of them.

The installation is failing because binaries aren't available for your platform, and you're missing the python development headers needed to install the cffi module. I _think_ `def install python3-devel` should install them.

shahab5, 2016-04-15:
Thanks it worked!!!

`pip2 uninstall yotta`   removed existing yotta

Installed development headers and yotta using,
`def install python3-devel`  
`pip install -U yotta`

closed 2016-04-15 by shahab5