ARMmbed/yotta #767: yotta search --limit 1000 target error

qymab, 2016-04-17:
using latest yotta.app for OSX running on Capitan.
```
Traceback (most recent call last):
  File "/Applications/yotta.app/Contents/Resources/workspace/bin/yotta", line 4, in <module>
    yotta.main()
  File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/yotta/main.py", line 239, in main
    status = args.command(args, following_args)
  File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/yotta/search.py", line 53, in execCommand
    for result in registry_access.search(query=args.query, keywords=args.kw, registry=args.registry):
  File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/yotta/lib/registry_access.py", line 679, in search
    response = requests.get(url, headers=headers, params=params)
  File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/requests/api.py", line 69, in get
    return request('get', url, params=params, **kwargs)
  File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/requests/api.py", line 50, in request
    response = session.request(method=method, url=url, **kwargs)
  File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/requests/sessions.py", line 471, in request
    resp = self.send(prep, **send_kwargs)
  File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/requests/sessions.py", line 579, in send
    r = adapter.send(request, **kwargs)
  File "/Applications/yotta.app/Contents/Resources/workspace/lib/python2.7/site-packages/requests/adapters.py", line 430, in send
    raise SSLError(e, request=request)
requests.exceptions.SSLError: ("bad handshake: Error([('SSL routines', 'ssl3_get_server_certificate', 'certificate verify failed')],)",)
```

autopulated, 2016-04-18:
Please make sure you have the latest certifi module installed: `pip install -U certifi`

qymab, 2016-04-18:
using `pip install -U certifi` I've got this 
`Collecting certifi
  Using cached certifi-2016.2.28-py2.py3-none-any.whl
Installing collected packages: certifi
Successfully installed certifi-2016.2.28
You are using pip version 7.1.2, however version 8.1.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.`

and then I'm trying with `pip install --upgrade pip' command`
It's working

closed 2016-04-18 by qymab

autopulated, 2016-04-18:
👍

I think this could have happened if you already had an old version of certifi installed before installing yotta with `pip install yotta`, possibly we should suggest installing with `pip install -U yotta`