ARMmbed/yotta #770: yt login gives Login timeout

janjongboom, 2016-04-20:
From the [forums](https://developer.mbed.org/questions/68753/yt-login-got-login-timeout/):

Q:when I input : yt login in the command line I get the respond from command line :

press enter to continue your browser will open to complete the login

then I login in the browser login page and get the follow page indication :

Woohoo!

Thanks for logging in - you're nearly done.

You can close this window and return to your console to check the login has been successful. tion If you're not logged in please run yotta login again.

and then I close the web page and return to the command line ,wait for about 5 mintes ,then it respond with login timeout error .

why this happen? thanks!

autopulated, 2016-04-20:
`yotta login --github` works for me.

`yotta -vvv login` might have some useful information to help debug

thegecko, 2017-03-28:
@janjongboom Is this problem still happening?

janjongboom, 2017-03-29:
@thegecko No clue, it was reported on the forums.