ARMmbed/yotta #776: absolute path on windows+cygwin make things complex

RdaZhongyao, 2016-05-05:
Environment: Windows + Cygwin
Tools: python/pip/cmake/ninja on cygwin, arm-none-eabi-gcc.exe on windows.

"yotta build" generates build rules, while most file has absolute path, such as : 
`build ym/dlmalloc/source/CMakeFiles/dlmalloc.dir/cygdrive/e/mbed/blinky/yotta_modules/dlmalloc/source/dlmalloc.c.o: C_COMPILER__dlmalloc /cygdrive/e/mbed/blinky/yotta_modules/dlmalloc/source/dlmalloc.c | yotta_config.h`

When compiling with arm-none-eabi-gcc.exe, these absolute path can not be recognized, only E:/mbed/blinky/... is valid.

Question: For yotta, why not generate relative path to compile? Is absolute path redundant?

BlackstoneEngineering, 2016-05-05:
@RdaZhongyao 
Cygwin is not supported as part of yotta. Cygwin munges the system path. Please use the system CMD tool.

RdaZhongyao, 2016-05-06:
@BlackstoneEngineering 
yotta docs does have description related to Cygwin, see : [yottadocs.mbed.com](http://yottadocs.mbed.com/#dependencies-linux) .
With this instruction, problem occurs as I told before.

The system CMD tool of Windows is OK. Actually, when tools (python/pip/cmake/ninja/yotta/gcc) are all from Windows platform, there is no problem. Because they use the same path format: "C:\mbed\test\".

However, at the point of using absolute path everywhere, yotta is not wise.

autopulated, 2016-05-06:
When using cygwin you need to use a compiler that understands cygwin paths (installed from the cygwin package manager)

BlackstoneEngineering, 2016-05-06:
The issue is actually within yotta, when yotta creates paths is is created based on system defaults. Cygwin uses `/` where windows uses `\`, also cygwin prepends things with `/cygdrive/`, there is actually a tool to convert between the two called [`cygpath`](http://cygwin-lite.sourceforge.net/html/cygpath.html). The fix is to have yotta detect if its running in cygwin, and if so create the paths and then run them through `cygpath` before using them. This creates significant overhead though and it was ultimately decided that we would not directly support Cygwin, instead the user is encouraged to use the native command line utility on their system. Thats Terminal on OSX, and CMD on Windows.

autopulated, 2016-05-06:
@BlackstoneEngineering That's incorrect. The paths in question are correctly emitted as /cygdrive/whatever paths by yotta and by CMake running under cygwin (as long as yotta has been installed with cygwin's python, which actually reports itself as a 'unix' platform, hence the / in the output paths), and as long as the compiler that CMake is driving can interpret /cygdrive paths (i.e. the compiler itself has been compiled for cygwin).

That said, the cygwin installation docs are probably not enough to get things working without other knowledge.

BlackstoneEngineering, 2016-05-06:
@autopulated my reccomendations come direclty from using cygwin for the last 4 years and trying to get yotta to run with Cygwin unsuccessfully.

RdaZhongyao, 2016-05-07:
Based on the previous discussion, we know that yotta doesn't support tools mixture that from different platform. Because they cannot understand the file path eachother. 

To set up an environment that combines Windows and Cygwin, two ways is given below.
Tips: Make sure the tools (python/pip/cmake/ninja/yotta/gcc) are from the same platform.

1) Tools all from Windows, Cygwin just as a SHELL.
Install python/pip/cmake/ninja/yotta/gcc tools by [yotta Windows installer](https://github.com/ARMmbed/yotta_windows_installer/releases). Add tool path to Cygwin PATH environment variable. While building, the generated path like this:
`build ym/dlmalloc/source/CMakeFiles/dlmalloc.dir/E_/mbed/blinky2/yotta_modules/dlmalloc/source/dlmalloc.c.o: C_COMPILER__dlmalloc E$:/mbed/blinky2/yotta_modules/dlmalloc/source/dlmalloc.c | yotta_config.h`

2) Tools all from Cygwin.
Install python/cmake with Cygwin Package Manager; Install pip with [get-pip.py](https://bootstrap.pypa.io/get-pip.py); Install yotta by "pip install yotta"; Install ninja with [source code](https://github.com/ninja-build/ninja) ([instruction](https://ninja-build.org/)); Install [arm-none-eabi-gcc](https://launchpad.net/gcc-arm-embedded) with source code. While building, the generated path like this:
`build ym/dlmalloc/source/CMakeFiles/dlmalloc.dir/cygdrive/e/mbed/blinky/yotta_modules/dlmalloc/source/dlmalloc.c.o: C_COMPILER__dlmalloc /cygdrive/e/mbed/blinky/yotta_modules/dlmalloc/source/dlmalloc.c | yotta_config.h`

Maybe yotta docs need to upgrade for now. :-)

One thing is still confusing me, is it possible that yotta generates relative path to support the mixture of tools that from Windows and Cygwin ?