ARMmbed/yotta #778: Login for downloading modules/targets/etc

Overdrivr, 2016-05-26:
Why is it required to login for performing any action that interacts with the registry ?

I feel this sort of requirement can make potential users not bother to create an mbed account and leave yotta early. 

This is the first time a package manager is enforcing something like this, and as a potential yotta developper, this is a clear deal-breaker for contributing.

Hope you can shed some light on this.

autopulated, 2016-05-26:
There was a requirement to collect the email addresses of everyone using yotta for (IMHO short-sighted) marketing purposes.

koko-ng, 2016-07-26:
I don't understand why we need such a package manager for c...Make, Cmake, etc. are great.

thegecko, 2017-04-24:
We have removed the need to log in to yotta for read-only access.

closed 2017-04-24 by thegecko