ARMmbed/yotta #782: Yotta installer

rajbatra, 2016-06-29:
I tried the yotta windows installer but it is not able to run yt init after going through run yotta.bat i observed that it is lookig for /workspace/scripts/activate however no such location exists . Is there a problem with installer

BlackstoneEngineering, 2016-06-29:
Have you tried running the installer with admin privileges?

AoifeOS, 2016-07-12:
I had the same problem and running with admin privileges didn't work.
rajbatra, did you find a solution?

BlackstoneEngineering, 2016-07-13:
Can y'all provide more details on your systems? Which version of windows are you running?

AoifeOS, 2016-07-13:
I'm running Windows 7 64 bit.

BlackstoneEngineering, 2016-07-13:
Can you please tell me what the contents of `C:\yotta` looks like?
If `c:\yotta` doesnt exist try creating the folder and then re-running the yotta installer.

anujdutt9, 2016-07-21:
Hey, use these steps to make it work:
So the steps are:
1. Download yotta installer. Ref.: http://yottadocs.mbed.com/#installing-on-windows
2. Install all features except mbed-serial for now.
3. Install python that comes with that.
4. Open Environment variables in windows.
5. In "YOTTA_PATH", put path of all files that you have in your yotta folder. example:

C:\yotta;C:\yotta\gcc\bin;C:\yotta\python;C:\yotta\python\Scripts;C:\yotta\cmake-3.4.0-rc3-win32-x86\bin

6.In "YOTTA_INSTALL_PATH" , put:

C:\yotta if your yotta folder is in C

7.In "PATH" append it as:

C:\yotta\mercurial;C:\yotta\python;C:\yotta and so on.

8.Now refresh and click on "RUN YOTTA" shortcut formed on your screen. It works.

I confirmed the steps by installing yotta 3-4 times. It works every time :slight_smile:
I'll soon post a video of all the steps on youtube.com. Hope this helps.