ARMmbed/yotta #783: Yotta manual install issue

Areontar, 2016-07-12:
Hi, I am not sure if this is the right venue for this, but it seems that cmake 3.6 is not compatible with the latest yotta. 3.5.2 works fine.  Might good to add that to instruction.

autopulated, 2016-07-12:
Is the problem related to this: https://github.com/ARMmbed/target-mbed-gcc/pull/29? / what's the error?

jacobrosenthal, 2016-08-05:
For osx users until this is fixed, I went back to the last version I could find in homebrew
```
brew uninstall cmake
brew tap homebrew/versions
brew install homebrew/versions/cmake31
```
BTW The error I get is
```
Call Stack (most recent call first):
  /Users/jacobrosenthal/Downloads/ble-example/BLE_LED/build/sticker-gcc/toolchain.cmake:9 (include)
  /Users/jacobrosenthal/Downloads/ble-example/BLE_LED/build/sticker-gcc/CMakeFiles/3.6.0/CMakeSystem.cmake:6 (include)
  /Users/jacobrosenthal/Downloads/ble-example/BLE_LED/build/sticker-gcc/CMakeFiles/CMakeTmp/CMakeLists.txt:3 (project)

-- Check for working C compiler: /usr/local/bin/arm-none-eabi-gcc -- broken
CMake Error at /usr/local/Cellar/cmake/3.6.0_1/share/cmake/Modules/CMakeTestCCompiler.cmake:61 (message):
  The C compiler "/usr/local/bin/arm-none-eabi-gcc" is not able to compile a
  simple test program.

  It fails with the following output:

   Change Dir: /Users/jacobrosenthal/Downloads/ble-example/BLE_LED/build/sticker-gcc/CMakeFiles/CMakeTmp

  

  Run Build Command:"/usr/local/bin/ninja" "cmTC_6f45c"

  [1/2] Building C object CMakeFiles/cmTC_6f45c.dir/testCCompiler.c.o

  [2/2] Linking C executable cmTC_6f45c

  FAILED: cmTC_6f45c

  : && /usr/local/bin/arm-none-eabi-gcc -fno-exceptions -fno-unwind-tables
  -Wl,--gc-sections -Wl,--sort-common -Wl,--sort-section=alignment
  -Wl,-wrap,main -Wl,-u,_printf_float -mcpu=cortex-m0 -mthumb
  -T"/Users/jacobrosenthal/Downloads/ble-example/BLE_LED/yotta_targets/nordic-nrf51822-gcc-unofficial-fota-custom/CMake/../ld/NRF51822_16K_S130.ld"
  -Wl,-Map,cmTC_6f45c.map -Wl,--start-group
  CMakeFiles/cmTC_6f45c.dir/testCCompiler.c.o -lm -lc -lgcc -lm -lc -lgcc
  -Wl,--end-group --specs=nano.specs -o cmTC_6f45c && :

  
  /usr/local/Cellar/gcc-arm-none-eabi-48/20140805/bin/../lib/gcc/arm-none-eabi/4.8.4/../../../../arm-none-eabi/bin/ld:
  warning: cannot find entry symbol Reset_Handler; defaulting to
  000000000001c000

  
  /usr/local/Cellar/gcc-arm-none-eabi-48/20140805/bin/../lib/gcc/arm-none-eabi/4.8.4/../../../../arm-none-eabi/lib/armv6-m/libc_s.a(lib_a-sbrkr.o):
  In function `_sbrk_r':

  sbrkr.c:(.text._sbrk_r+0xc): undefined reference to `_sbrk'

  collect2: error: ld returned 1 exit status

  ninja: build stopped: subcommand failed.

  

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:126 (project)

-- Configuring incomplete, errors occurred!
See also "/Users/jacobrosenthal/Downloads/ble-example/BLE_LED/build/sticker-gcc/CMakeFiles/CMakeOutput.log".
See also "/Users/jacobrosenthal/Downloads/ble-example/BLE_LED/build/sticker-gcc/CMakeFiles/CMakeError.log".
error: command ['cmake', '-D', 'CMAKE_BUILD_TYPE=RelWithDebInfo', '-G', 'Ninja', '.'] failed
```

LiyouZhou, 2017-06-13:
I have experienced this as well on a mac sierra. Effectively mbed-os 3 targets are broken because of this.

akosthekiss, 2018-05-17:
I know that this is a really old issue but someone might still be interested in a workaround. Creating an executable script named `cmake` like this:
```sh
#!/bin/bash
$WCMAKE -DCMAKE_C_COMPILER_WORKS=TRUE $@
```
and calling yotta like this:
```sh
export WCMAKE=`which cmake`
export PATH=`pwd`:$PATH
yotta build
```
solves the problem. By wrapping cmake, this ensures that `-DCMAKE_C_COMPILER_WORKS=TRUE` gets passed to it on top of whatever command line options yotta sets (and this option disables the compiler check in cmake). This workaround has been put in use at https://github.com/jerryscript-project/jerryscript/commit/53bd845d9fd45fa58ac2f47353687ac825cacf0c .

Of course, if a workaround is not good enough, yotta itself should be patched up somewhere here: https://github.com/ARMmbed/yotta/blob/master/yotta/lib/target.py#L494 . Something like:
```diff
- cmd = ['cmake', '-D', 'CMAKE_BUILD_TYPE=%s' % build_type, '-G', args.cmake_generator, '.']
+ cmd = ['cmake', '-D', 'CMAKE_BUILD_TYPE=%s' % build_type, '-D', 'CMAKE_C_COMPILER_WORKS=TRUE', '-G', args.cmake_generator, '.']
```