ARMmbed/yotta #784: 'yotta build' fails with error "command ['ninja'] failed"

AoifeOS, 2016-07-12:
I am using Ubuntu 14.04 LTS and trying to use frdm-k64f-gcc as my target.
All I want to do is compile and run blinky. This is the last section of the output when I run:

`yotta build`

```
/home/nx/example-mbedos-blinky/yotta_modules/greentea-client/source/test_env.cpp:19:18: fatal error: cstdio: No such file or directory
 #include <cstdio>
                  ^
compilation terminated.
[12/143] Building C object ym/dlmalloc...a_modules/dlmalloc/source/dlmalloc.c.o
ninja: build stopped: subcommand failed.
error: command ['ninja'] failed
```

No binaries are created when I run this.
Any help would be appreciated!

autopulated, 2016-07-12:
This is a build error due to a problem with the source files you're compiling: in this case it looks like the error happened when compiling the source for the "greentea-client" module, so a good place to start would be here: https://github.com/armmbed/greentea-client

AoifeOS, 2016-07-12:
I tried using these two commands

pip install mbed-greentea --upgrade

pip install "mbed-greentea<0.2.0" --upgrade

and I tried to build it again after that and I got the same result as before.

autopulated, 2016-07-12:
cc @PrzemekWirkus @thegecko @jaustin

bogdanm, 2016-07-12:
Are you using https://launchpad.net/gcc-arm-embedded? I've seen similar errors for other distributions of the GCC ARM toolchain.

bogdanm, 2016-07-12:
(also, @autopulated is right, this error is not in `yotta` itself).

PrzemekWirkus, 2016-07-12:
@AoifeOS Hi! Errors related to `example-mbedos-blinky/yotta_modules/greentea-client` are not in this [Greentea](https://github.com/ARMmbed/greentea). They come from source code inside blinky example!

PrzemekWirkus, 2016-07-12:
My steps to reproduce Blinky example:

```
przemek@e100452:~$ mkdir blinky
```
```
przemek@e100452:~$ cd blinky/
```
```
przemek@e100452:~/blinky$ git clone https://github.com/ARMmbed/example-mbedos-blinky.git
Cloning into 'example-mbedos-blinky'...
remote: Counting objects: 85, done.
remote: Total 85 (delta 0), reused 0 (delta 0), pack-reused 85
Unpacking objects: 100% (85/85), done.
Checking connectivity... done.
```
```
przemek@e100452:~/blinky$ cd example-mbedos-blinky/
```
```
przemek@e100452:~/blinky/example-mbedos-blinky$ yt target frdm-k64f-gcc
```
```
info: get versions for frdm-k64f-gcc
info: download frdm-k64f-gcc@2.0.0 from the public module registry
info: get versions for kinetis-k64-gcc
info: download kinetis-k64-gcc@2.2.0 from the public module registry
info: get versions for mbed-gcc
info: download mbed-gcc@1.2.2 from the public module registry
```
```
przemek@e100452:~/blinky/example-mbedos-blinky$ yt up
```
```
info: get versions for frdm-k64f-gcc
info: get versions for kinetis-k64-gcc
info: get versions for mbed-gcc
info: get versions for mbed-drivers
info: download mbed-drivers@1.5.0 from the public module registry
info: get versions for mbed-hal
info: download mbed-hal@1.3.0 from the public module registry
info: get versions for cmsis-core
info: download cmsis-core@1.2.0 from the public module registry
info: get versions for ualloc
info: download ualloc@1.3.0 from the public module registry
info: get versions for minar
info: download minar@1.3.0 from the public module registry
info: get versions for core-util
info: download core-util@1.8.0 from the public module registry
info: get versions for compiler-polyfill
info: download compiler-polyfill@1.3.0 from the public module registry
info: get versions for greentea-client
info: download greentea-client@1.1.0 from the public module registry
info: get versions for mbed-hal-freescale
info: download mbed-hal-freescale@1.1.0 from the public module registry
info: get versions for mbed-hal-ksdk-mcu
info: download mbed-hal-ksdk-mcu@1.3.0 from the public module registry
info: get versions for uvisor-lib
info: download uvisor-lib@2.1.2 from the public module registry
info: get versions for mbed-hal-k64f
info: download mbed-hal-k64f@1.3.0 from the public module registry
info: get versions for cmsis-core-freescale
info: download cmsis-core-freescale@1.1.0 from the public module registry
info: get versions for cmsis-core-k64f
info: download cmsis-core-k64f@1.1.0 from the public module registry
info: get versions for mbed-hal-frdm-k64f
info: download mbed-hal-frdm-k64f@1.1.1 from the public module registry
info: get versions for dlmalloc
info: download dlmalloc@1.1.0 from the public module registry
info: get versions for minar-platform
info: download minar-platform@1.1.0 from the public module registry
info: get versions for minar-platform-mbed
info: download minar-platform-mbed@1.3.0 from the public module registry
```
```
przemek@e100452:~/blinky/example-mbedos-blinky$ yt build
```
```
info: generate for target: frdm-k64f-gcc 2.0.0 at /home/przemek/blinky/example-mbedos-blinky/yotta_targets/frdm-k64f-gcc
GCC version is: 4.9.3
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/przemek/blinky/example-mbedos-blinky/build/frdm-k64f-gcc
[58/143] Building CXX object ym/mbed-drivers/source/CMakeFiles/mbed-drivers.dir/home/przemek/blinky/example-mbedos-blinky/yotta_modules/mbed-drivers/source/test_env.cpp.o
In file included from /home/przemek/blinky/example-mbedos-blinky/yotta_modules/mbed-drivers/source/test_env.cpp:17:0:
/home/przemek/blinky/example-mbedos-blinky/yotta_modules/mbed-drivers/mbed-drivers/test_env.h:20:2: warning: #warning mbed-drivers/test_env.h is deprecated. Please use greentea-client/test_env.h instead. [-Wcpp]
 #warning mbed-drivers/test_env.h is deprecated. Please use greentea-client/test_env.h instead.
  ^
[63/143] Building C object ym/core-util/source/CMakeFiles/core-util.dir/home/przemek/blinky/example-mbedos-blinky/yotta_modules/core-util/source/critical.c.o
/home/przemek/blinky/example-mbedos-blinky/yotta_modules/core-util/source/critical.c: In function 'core_util_critical_section_exit':
/home/przemek/blinky/example-mbedos-blinky/yotta_modules/core-util/source/critical.c:65:18: warning: unused variable 'primask' [-Wunused-variable]
         uint32_t primask = __get_PRIMASK(); // get the current interrupt enabled state
                  ^
[143/143] Linking CXX executable source/example-mbedos-blinky
```

AoifeOS, 2016-07-13:
@PrzemekWirkus I followed your steps to reproduce blinky from scratch and got the same error again.

I have a different version of arm-none-eabi-gcc.
`nx@nx-OptiPlex-390:~/blinky/example-mbedos-blinky$ arm-none-eabi-gcc --version` 
```
arm-none-eabi-gcc (4.8.2-14ubuntu1+6) 4.8.2
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

bogdanm, 2016-07-13:
Try to install and use GCC ARM embedded from https://launchpad.net/gcc-arm-embedded. I suspect that might fix your problems.

PrzemekWirkus, 2016-07-13:
@AoifeOS LIke @bogdanm have said. You need to isntall GCC ARM Embedded Compiler from Launchpad. One you have might not work correctly.

PrzemekWirkus, 2016-07-13:
@AoifeOS Here is detailed instruction: http://yottadocs.mbed.com/#linux-cross-compile

AoifeOS, 2016-07-13:
It works now! Thanks so much guys!

closed 2017-03-28 by thegecko

Lithimlin, 2018-03-28:
I'm having the same problem, though I'm trying to use the microbit libraries and the solution above did not work for me