ARMmbed/yotta #785: 'yotta' is not recognized as an internal or external command - after successful Windows install

georgeedwards, 2016-07-19:
I ran the Windows installer (v023) which I downloaded from the link in the tutorial. After the install executable ran, it said it had completed successfully and the yotta icon appeared on my desktop. However, when I run the `yotta` command in the terminal that that icon opens I get the following error:

>'yotta' is not recognized as an internal or external command,
operable program or batch file.

[![enter image description here][1]][1]
 [1]: http://i.stack.imgur.com/F3i7t.png

I am running windows 10.

anujdutt9, 2016-07-21:
Hey, I got it working using following steps:
So the steps are:
1. Download yotta installer. Ref.: http://yottadocs.mbed.com/#installing-on-windows
2. Install all features except mbed-serial for now.
3. Install python that comes with that.
4. Open Environment variables in windows.
5. In "YOTTA_PATH", put path of all files that you have in your yotta folder. example:

C:\yotta;C:\yotta\gcc\bin;C:\yotta\python;C:\yotta\python\Scripts;C:\yotta\cmake-3.4.0-rc3-win32-x86\bin

6.In "YOTTA_INSTALL_PATH" , put:

C:\yotta if your yotta folder is in C

7.In "PATH" append it as:

C:\yotta\mercurial;C:\yotta\python;C:\yotta and so on.

8.Now refresh and click on "RUN YOTTA" shortcut formed on your screen. It works.

I confirmed the steps by installing yotta 3-4 times. It works every time :slight_smile:
Hope this helps.

georgeedwards, 2016-07-31:
@anujdutt9 I have uninstalled everything and tried your steps twice, and I still am not getting this to work, see these screenshots below in case something is wrong there. However, when the installer pulls up a terminal window (I believe it is when python is being installed?), there were a load of red error messages, but the window disappears before you can read them. There was no record of this in the final installer. 

[![enter image description here][1]][1]

[![enter image description here][2]][2]

[![enter image description here][3]][3]

  [1]: http://i.stack.imgur.com/b4BHA.png
  [2]: http://i.stack.imgur.com/rI43i.png
  [3]: http://i.stack.imgur.com/EoqW0.png

BlackstoneEngineering, 2016-08-01:
@georgeedwards what OS and what editor are you trying to use this in? Currently yotta installs into a virtual environment seperate from your global space. You may need to try and run the installer with admin privelidges, or try and do `pip install -U yotta

georgeedwards, 2016-08-01:
@BlackstoneEngineering I am just trying this from the command line atm? I'll try and reinstall as Admin.

georgeedwards, 2016-08-01:
@BlackstoneEngineering Still no... However, here is the log:

```
C:\Users\George>pip install -U yotta
You are using pip version 7.0.1, however version 8.1.2 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
Collecting yotta
Collecting argcomplete<2.0,>=0.8.0 (from yotta)
  Using cached argcomplete-1.4.1-py2.py3-none-any.whl
Collecting certifi>=2016.2.28 (from yotta)
  Using cached certifi-2016.2.28-py2.py3-none-any.whl
Collecting Jinja2<3,>=2.7.0 (from yotta)
  Using cached Jinja2-2.8-py2.py3-none-any.whl
Collecting semantic-version<3,>=2.3.1 (from yotta)
Collecting ntfsutils<0.2,>=0.1.3 (from yotta)
Collecting jsonschema<3.0,>=2.4.0 (from yotta)
  Using cached jsonschema-2.5.1-py2.py3-none-any.whl
Collecting cryptography>=0.8 (from yotta)
  Using cached cryptography-1.4-cp27-none-win32.whl
Collecting colorama<0.4,>=0.3 (from yotta)
  Using cached colorama-0.3.7-py2.py3-none-any.whl
Collecting requests[security]<3,>=2.4.3 (from yotta)
  Using cached requests-2.10.0-py2.py3-none-any.whl
Collecting PyGithub<2,>=1.25 (from yotta)
Collecting PyJWT<2.0,>=1.0 (from yotta)
  Using cached PyJWT-1.4.1-py2.py3-none-any.whl
Collecting jsonpointer<2.0,>=1.9 (from yotta)
  Using cached jsonpointer-1.10-py2-none-any.whl
Collecting valinor<1.0,>=0.0.0 (from yotta)
Collecting mbed-test-wrapper<2.0.0,>=0.0.3 (from yotta)
Collecting hgapi<2,>=1.7 (from yotta)
Collecting intelhex<3.0,>=2.0 (from yotta)
  Using cached intelhex-2.1-py2.py3-none-any.whl
Collecting pathlib<1.1,>=1.0.1 (from yotta)
Collecting MarkupSafe (from Jinja2<3,>=2.7.0->yotta)
  Using cached MarkupSafe-0.23.tar.gz
Collecting setuptools (from ntfsutils<0.2,>=0.1.3->yotta)
  Using cached setuptools-25.1.1-py2.py3-none-any.whl
Collecting functools32 (from jsonschema<3.0,>=2.4.0->yotta)
Collecting enum34 (from cryptography>=0.8->yotta)
  Using cached enum34-1.1.6-py2-none-any.whl
Collecting ipaddress (from cryptography>=0.8->yotta)
  Using cached ipaddress-1.0.16-py27-none-any.whl
Collecting pyasn1>=0.1.8 (from cryptography>=0.8->yotta)
  Using cached pyasn1-0.1.9-py2.py3-none-any.whl
Collecting six>=1.4.1 (from cryptography>=0.8->yotta)
  Using cached six-1.10.0-py2.py3-none-any.whl
Collecting idna>=2.0 (from cryptography>=0.8->yotta)
  Using cached idna-2.1-py2.py3-none-any.whl
Collecting cffi>=1.4.1 (from cryptography>=0.8->yotta)
  Using cached cffi-1.7.0-cp27-none-win32.whl
Collecting pyOpenSSL>=0.13 (from requests[security]<3,>=2.4.3->yotta)
  Using cached pyOpenSSL-16.0.0-py2.py3-none-any.whl
Collecting ndg-httpsclient (from requests[security]<3,>=2.4.3->yotta)
Collecting pyelftools==0.23 (from valinor<1.0,>=0.0.0->yotta)
Collecting pyOCD<1.0,>=0.3 (from valinor<1.0,>=0.0.0->yotta)
Collecting project-generator<0.9.0,>=0.8.0 (from valinor<1.0,>=0.0.0->yotta)
Collecting pyyaml<4,>=3 (from valinor<1.0,>=0.0.0->yotta)
  Using cached PyYAML-3.11.zip
Collecting pycparser (from cffi>=1.4.1->cryptography>=0.8->yotta)
Collecting pywinusb>=0.4.0 (from pyOCD<1.0,>=0.3->valinor<1.0,>=0.0.0->yotta)
Collecting argparse (from project-generator<0.9.0,>=0.8.0->valinor<1.0,>=0.0.0->yotta)
  Using cached argparse-1.4.0-py2.py3-none-any.whl
Collecting xmltodict (from project-generator<0.9.0,>=0.8.0->valinor<1.0,>=0.0.0->yotta)
Collecting project-generator-definitions<0.3.0,>=0.2.2 (from project-generator<0.9.0,>=0.8.0->valinor<1.0,>=0.0.0->yotta)
Installing collected packages: argcomplete, certifi, MarkupSafe, Jinja2, semantic-version, setuptools, ntfsutils, functools32, jsonschema, enum34, ipaddress, pyasn1, six, idna, pycparser, cffi, cryptography, colorama, pyOpenSSL, ndg-httpsclient, requests, PyGithub, PyJWT, jsonpointer, pyelftools, pywinusb, intelhex, pyOCD, argparse, xmltodict, pyyaml, project-generator-definitions, project-generator, valinor, mbed-test-wrapper, hgapi, pathlib, yotta
  Running setup.py install for MarkupSafe
  Found existing installation: setuptools 16.0
    Uninstalling setuptools-16.0:
      Successfully uninstalled setuptools-16.0
[... 10 lines cut ...]
    copying lib\yaml\cyaml.py -> build\lib.win32-2.7\yaml
    copying lib\yaml\dumper.py -> build\lib.win32-2.7\yaml
    copying lib\yaml\emitter.py -> build\lib.win32-2.7\yaml
    copying lib\yaml\error.py -> build\lib.win32-2.7\yaml
    copying lib\yaml\events.py -> build\lib.win32-2.7\yaml
    copying lib\yaml\loader.py -> build\lib.win32-2.7\yaml
    copying lib\yaml\nodes.py -> build\lib.win32-2.7\yaml
    copying lib\yaml\parser.py -> build\lib.win32-2.7\yaml
    copying lib\yaml\reader.py -> build\lib.win32-2.7\yaml
    copying lib\yaml\representer.py -> build\lib.win32-2.7\yaml
    copying lib\yaml\resolver.py -> build\lib.win32-2.7\yaml
    copying lib\yaml\scanner.py -> build\lib.win32-2.7\yaml
    copying lib\yaml\serializer.py -> build\lib.win32-2.7\yaml
    copying lib\yaml\tokens.py -> build\lib.win32-2.7\yaml
    copying lib\yaml\__init__.py -> build\lib.win32-2.7\yaml
    running build_ext
    creating build\temp.win32-2.7
    creating build\temp.win32-2.7\Release
    checking if libyaml is compilable
    error: [Error 2] The system cannot find the file specified

    ----------------------------------------
Command "C:\yotta\python\python.exe -c "import setuptools, tokenize;__file__='c:\\users\\george\\appdata\\local\\temp\\pip-build-ntry_n\\pyyaml\\setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record c:\users\george\appdata\local\temp\pip-x5wxzv-record\install-record.txt --single-version-externally-managed --compile" failed with error code 1 in c:\users\george\appdata\local\temp\pip-build-ntry_n\pyyaml

```

BlackstoneEngineering, 2016-08-01:
update your pip, `pip install -U pip`, then try again.
Also try updating setup tools, `pip install -U setuptools`
Also, just for funsies, try doing this : `pip install -U requests[security]` 

That last one is part of my magic voodoo dance that occasionally fixes random things.

georgeedwards, 2016-08-01:
@BlackstoneEngineering  I am afraid still no joy, [here](https://gist.github.com/georgeedwards/99ae548c102aea73e8fd94324ca07621) you can see the log of me running those commands and then retrying the yotta install. The pip install threw some exceptions but then seemed to have worked?

georgeedwards, 2016-08-04:
Firstly, I would really appreciate some more assistance with getting this up and running because there is obviously some overlooked aspect of my system causing me grief.  

Secondly, I think this (and the numerous other questions and issues) raises the point that the current installer / installation steps aren't sufficient. I have just come across [Vagrant](https://www.vagrantup.com/) which seems to be an open-source CLI to automate development environments, specifically to avoid issues like this.

biomindr, 2016-08-04:
Hi, 

It seems as there is a problem with the installer on Windows 8.1 as well since I'm experiencing the same problem. When I execute run_yotta.bat, it results in an error 'cmd' is not recognized as an internal or external command, operable program or batch file". So the last command line of the batch file is not executed: cmd /K "cd %HOMEPATH% & %YOTTA_INSTALL_LOCATION%\workspace\Scripts\activate"

Moreover when I look in the path %YOTTA_INSTALL_LOCATION% I do not see any folder called workspace?

Here are my paths following the previous intructions, except for YOTTA_INSTALL_PATH I did use YOTTA_INSTALL_LOCATION :

C:\yotta>echo %YOTTA_INSTALL_PATH%
%YOTTA_INSTALL_PATH%

C:\yotta>echo %PATH%
C:\yotta;C:\yotta\gcc\bin;C:\yotta\python;C:\yotta\python\Scripts;C:\yotta\cmake
-3.4.0-rc3-win32-x86\bin;C:\Program Files (x86)\yotta\python\;C:\Program Files (
x86)\yotta\python\Scripts;C:\Program Files\Java\jre1.8.0_60\bin;C:\bin\apache-an
t-1.9.6\bin;C:\Program Files (x86)\Android\android-sdk\platform-tools;C:\Program
 Files (x86)\Android\android-sdk\tools; C:\bin\Android_NDK\android-ndk-r10e;C:\M
inGW\bin;C:\bin\Android_NDK\android-ndk-r10e;C:\Program Files\Git\cmd;C:\yotta\m
ercurial;C:\yotta\python;C:\yotta

C:\yotta>echo %YOTTA_INSTALL_LOCATION%
C:\yotta

C:\yotta>echo %YOTTA_PATH%
C:\yotta;C:\yotta\gcc\bin;C:\yotta\python;C:\yotta\python\Scripts;C:\yotta\cmake
-3.4.0-rc3-win32-x86\bin

I hope to have some feedback soon since I need to get my environment up and running by tomorrow. Thanks in advance.

BlackstoneEngineering, 2016-08-05:
Srry guys, I don't know what the problem is nor do I have a windows 8.1 machine to test against. My suggestion would be to use the manual installation steps listed [here](http://yottadocs.mbed.com/#installing) 

Leme know if you get it working or not.

georgeedwards, 2016-08-05:
OK, I have downloaded Vagrant and provisioned a VirtualBox base VM running the vagrant `hashicorp/precise64` which is "A standard Ubuntu 12.04 LTS 32-bit box". I ran the Linux:

>sudo apt-get update && sudo apt-get install python-setuptools  cmake build-essential ninja-build python-dev libffi-dev libssl-dev && sudo easy_install pip

command fine and then `pip install yotta`, however,  that ends with the same error message I was getting before:

>Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-E4sLwI/semantic-version/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-b0T9KR-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-E4sLwI/semantic-version/

Why is this happening on a clean VM? I tried repeating the pip install and also going through the chown procedure outlined in the docs, I always get the same result...

anujdutt9, 2016-08-05:
This is because you cannot install  pip  on vagrant. I have already tried
all methods. Although using a Ubuntu machine directly or hosting ubuntu via
virtualbox can solve your problem.
I have tried these instructions with Windows 7 and Windows 10 and both work
fine.

georgeedwards, 2016-08-05:
@anujdutt9 Well If I run : 

>vagrant@precise64:~$ pip --version

then I get:

>pip 8.1.2 from /usr/local/lib/python2.7/dist-packages/pip-8.1.2-py2.7.egg (python 2.7)

Whihc suggests it has installed successfully no?

anujdutt9, 2016-08-05:
Yes agreed, but yotta dosen't works well with vagrant. I already tried
that. Thats why I moved to ubuntu on Virtualbox.

biomindrSA, 2016-08-05:
And I am currently installing Virtual Box to use Ubuntu as well. But If Windows is supported, the installer should be updated.

BlackstoneEngineering, 2016-08-08:
FYI there are known issues with the Ubuntu 14.04 LTS distro, primiarily related to it has a suuuuuuper old version of pip and python. I suggest using a 15.x image and updating everything before you get going. Make sure your pip is at least version 8.x, otherwise you *will* have problems.

georgeedwards, 2016-08-10:
Is there anyone from the core team who is willing to help on this, because it seems that currently, a number of Windows users can't use this project for a reason which isn't yet a known issue. I am really keen to be able to play with this.

bridadan, 2016-08-10:
@georgeedwards this is potentially unrelated to your issue, but I noticed in your logs that its choking on installing the `pyyaml` package. I've had trouble installing this package in the past inside of a virtual environment before on Windows. I believe this package is actually built natively for the platform, which may be contributing to your issue.

In the past I've gotten around this by installing `pyyaml` to the global pip packages and allowing my virtual environment access to the system site packages. Though I think this might not be a valid solution in this case.

I did some searching and it looks like other people have had this problem as well. This Stack Overflow issue seems to be the most relevant in your case: http://stackoverflow.com/questions/33665181/how-to-install-pyyaml-on-windows-10. I'd try the accepted solutions instructions inside your virtual environment first. Just be sure to grab the right package. If you're running Python 2.7 on Windows 64 bit, then you'll want the `PyYAML-3.11-cp27-cp27m-win_amd64.whl` download I think.