ARMmbed/yotta #786: `yotta build` rebuilds everything every time

SimonSapin, 2016-07-22:
When I run `yotta build` for this module: https://github.com/lancaster-university/microbit-samples/ I get a bunch of output on the terminal that ends after a dozen seconds or so with:

```
[158/158] Linking CXX executable source/microbit-samples
```

This seems normal.

But when I run it again without touching any file it still takes a dozen seconds and produces all of the same output. I would expect the build system to reuse at least some of the generated files when the source files haven’t changed.

Am I doing something wrong?

autopulated, 2016-07-26:
Definitely doesn't sound right – probably the most interesting thing to see what's causing the re-build is the first thing that gets re-built.

If you build with `yotta build -- -v`, and compare the start of the output between a clean build and a re-build, are there any differences?

SimonSapin, 2016-07-26:
I… can’t seem to reproduce this now. After `yotta clean`, the second build finishes quickly with (some other output and) "ninja: no work to do."

So, uh, feel free to close.

autopulated, 2016-07-28:
👍 
It's possible it was caused by clock skew / files being created/copied from another machine with timestamps in the future.

closed 2017-03-28 by thegecko