ARMmbed/yotta #792: Tests under Yotta

plauche, 2017-01-05:
I'm currently working through writing unit tests for a few Yotta modules and I'm trying to figure out the best way to stub or mock out dependencies under Yotta. Has anyone else wrestled with this and found a good solution? The json configuration files do not appear to support replacing dependencies when building tests. I also don't see any flags set (by default) which indicate that the current module being built is a library with tests present.

BlackstoneEngineering, 2017-01-06:
we use greentea for all microcontroller testing. github.com/armmbed/greentea

plauche, 2017-01-06:
@BlackstoneEngineering I'll have to take a look at that. We aren't using yotta under mbed but it would be interesting to see how the tool works. Do you use it for unit testing and integration testing?

BlackstoneEngineering, 2017-01-06:
Yes, and its not tied to mbed. Greentea is based on Utest and really just adds some extra bits for automatically loading programs onto microcontrollers and returning the results across the serial channel.

thegecko, 2017-03-28:
@plauche, is this issue resolved?

plauche, 2017-03-28:
@thegecko, yes, we were able to come to a satisfactory solution using additional `.cmake` files under the test folders to get the linker to create function stubs.

closed 2017-03-29 by thegecko