ARMmbed/yotta #793: build url for login is invalid

petertorelli, 2017-01-24:
While building the helloyotta example, I'm prompted for a login, but the URL generates a "page not found" on GitHub:

```
yotta workspace) ~/proj/yotta/helloyotta $ yotta build
info: get versions for x86-osx-native
info: download x86-osx-native@0.0.8 from the public module registry

You need to log in to do this.
Press enter to continue.
Your browser will open to complete login.

waiting for response...
If you are unable to use a browser on this machine, please copy and paste this URL into a browser:
http://yottabuild.org/#login.... long url here
```

[image]

ilg-ul, 2017-01-24:
FYI: The last commit dates from May 20th, 2016. Apparently ARM, in its infinite wisdom, decided to cancel yotta development.

thegecko, 2017-01-24:
@petertorelli Which version of yotta are you using?

```bash
$ yt --version
```

The recent versions use a login url which begins `https://yotta.mbed.com/#login/...`. I've checked the login works as expected for `yotta 0.16.4`.

petertorelli, 2017-01-30:
@thegecko .. 0.8.3. Yikes, I suspect I am very out of date. I thought since I downloaded the OSX image from the mbed site that it was the latest. Thanks, I'll try this.

BlackstoneEngineering, 2017-01-30:
aye, the yotta installer for OSX is way out of date. No intention of updating it though. Would suggest you either update it in place using Pip or install it manually