ARMmbed/yotta #799: 'certificate verify failed' since today

sgk, 2017-03-25:
Yotta does not work since today. It worked fine yesterday. What happened to my environment or the server? The error is as follows:

```
error: connection error: ("bad handshake: Error([('SSL routines', 'ssl3_get_server_certificate', 'certificate verify failed')],)",)
```

My environment is yotta-0.17.2 and Python 2.7.12 on Ubuntu 16.04 LTS.

thegecko, 2017-03-25:
Thanks @sgk,

It seems an SSL certificate has expired, please bear with us while we get it renewed.

sgk, 2017-03-25:
Oh, I see. Thank you for keeping the system up and running.

fred260571, 2017-03-26:
Maybe you should have an unsafe option ie
yotta unsafe target bbc-microbit-classic-gcc
that will allow users NOT to use SSL
What about a simple md5sum check ?
All I need is to install an IDE/compiler to program my BBC microbit
I tried in vain for 8 hours yesterday , needless to say that I was/am pretty upset
This is the last time I use an mbed/yotta project
My next project will use an Arduino
Anyway thanks for your hard work

autopulated, 2017-03-26:
If you want to quickly hack around this, you could add something like this after the line that imports the request module in yotta/lib/registry_access.py:

```py
import requests

_old_get = requests.get
def newGet(*args, **kwargs):
    kwargs['verify'] = False
    return _old_get(*args, **kwargs)
requests.get = newGet

```
(untested, but you get the idea)

The SSL is there for good reason though – without it if you were using yotta on an open wifi network then anyone else would be able to intercept your requests to the registry and insert malicious responses (or worse if you were publishing a module: they would be able to insert malicious code into the published module)

thegecko, 2017-03-26:
Thanks for your patience with this issue. We believe we have fixed it, but I'll keep this issue open for a day or so just in case.
Apologies on the untimely resolution, there was an issue with our hosting provider.

closed 2017-03-27 by thegecko