ARMmbed/yotta #803: Semver incompatibility

jamesadevine, 2017-04-06:
Hi,

It seems the recent version has broken semantic version for any previous version, which we heavily use in our project, [microbit-dal](https://github.com/lancaster-university/microbit-dal).

We have had two new users on v18 who have reported this breakage: https://github.com/lancaster-university/microbit-dal/issues/282

Any help would be greatly appreciated :smile:

thegecko, 2017-04-06:
Hi @jamesadevine 
I don't think this is semver version issue, but a problem with the `+` characters in the tag names being used.

The method and regex's used to correctly support github URLs was changed in the recent release.

A fix would likely require a modification to the regex to add a `+` here:

https://github.com/ARMmbed/yotta/blob/master/yotta/lib/sourceparse.py#L60

closed 2017-04-06 by thegecko

jamesadevine, 2017-04-07:
@thegecko Thanks for the amazingly swift response and solution :smile:

thegecko, 2017-04-07:
nps

PR #804 (2017-04-06): Fixed support for tags containing '+'
Fixes #803 

@bogdanm @marcuschangarm @0xc0170