ARMmbed/yotta #819: Login link results in error after successful login

carlosperate, 2017-09-15:
Login into the link that Yotta sends you after `yt login` displays the normal login screen, after entering the user details correctly, it redirects to:
https://yottabuild.org/#error/Unexpected%20token%20%3C%20in%20JSON%20at%20position%200

And shows as:

[image]

Tried in a couple of different PCs running the same vagrant virtualbox set up with Ubuntu 16.04, installing over ssh. This is something that was working perfectly fine last week, so wondering if something is broken on the server side.

thegecko, 2017-09-15:
There hasn't been any changes. Can you try forcing a logout with `yotta logout` first?

autopulated, 2017-09-15:
It sounds like the API server has returned an html page instead of JSON – heroku temporarily down / reached some rate limit?

sgk, 2017-09-17:
Exactly same problem here. `yotta logout` does not change the result.

thegecko, 2017-09-17:
Thanks all for bringing this to our attention. We believe we have found the issue and will deploy a fix as soon as we can.

thegecko, 2017-09-17:
This issue has been fixed and verified. Apologies for any disruption.

closed 2017-09-17 by thegecko

carlosperate, 2017-09-18:
I can confirm this is working again now, thanks!