ARMmbed/yotta #822: Certificate verification failed

pestalella, 2017-12-02:
I just installed yotta on a Windows 10 laptop and I'm getting these errors when I try to set the build target.

`(workspace) C:\Users\pau\prog\microbit-samples>yt target bbc-microbit-classic-gcc
info: get versions for bbc-microbit-classic-gcc
From cffi callback <function _verify_callback at 0x0374B8B0>:
Traceback (most recent call last):
  File "c:\yotta\workspace\lib\site-packages\OpenSSL\SSL.py", line 313, in wrapper
    _lib.X509_up_ref(x509)
AttributeError: 'module' object has no attribute 'X509_up_ref'
warning: connection error: HTTPSConnectionPool(host='registry.yottabuild.org', port=443): Max retries exceeded with url: /targets/bbc-microbit-classic-gcc/versions (Caused by SSLError(SSLError("bad handshake: Error([('SSL routines', 'tls_process_server_certificate', 'certificate verify failed')],)",),)), retrying...
From cffi callback <function _verify_callback at 0x0374BAB0>:
Traceback (most recent call last):
  File "c:\yotta\workspace\lib\site-packages\OpenSSL\SSL.py", line 313, in wrapper
    _lib.X509_up_ref(x509)
AttributeError: 'module' object has no attribute 'X509_up_ref'
warning: connection error: HTTPSConnectionPool(host='registry.yottabuild.org', port=443): Max retries exceeded with url: /targets/bbc-microbit-classic-gcc/versions (Caused by SSLError(SSLError("bad handshake: Error([('SSL routines', 'tls_process_server_certificate', 'certificate verify failed')],)",),)), retrying...
`

If I try to run wget on registry.yottabuild.org:

`C:\Users\pau>"c:\Program Files (x86)\GnuWin32\bin\wget.exe" -d https://registry.yottabuild.org
SYSTEM_WGETRC = c:/progra~1/wget/etc/wgetrc
syswgetrc = c:\Program Files (x86)\GnuWin32/etc/wgetrc
DEBUG output created by Wget 1.11.4 on Windows-MinGW.

--2017-12-02 12:27:27--  https://registry.yottabuild.org/
Resolving registry.yottabuild.org... seconds 0.00, 184.73.200.185, 184.73.203.207, 184.73.247.198, ...
Caching registry.yottabuild.org => 184.73.200.185 184.73.203.207 184.73.247.198 23.21.206.136 204.236.236.192 23.23.118.21 23.21.55.239 23.21.222.104
Connecting to registry.yottabuild.org|184.73.200.185|:443... seconds 0.00, connected.
Created socket 556.
Releasing 0x001fb438 (new refcount 1).
Initiating SSL handshake.
SSL handshake failed.
Closed fd 556
Unable to establish SSL connection.
`

DarrellBailey, 2017-12-12:
There seems to be a dependency version issue. I solved the issue by using the following command:

`pip install pyopenssl yotta`

pestalella, 2017-12-12:
Nice! It worked. Thanks a lot for the help :)

closed 2017-12-12 by pestalella