ARMmbed/yotta #823: OpenSSL binding error on Ubuntu 16.04.3

sgk, 2017-12-03:
After installing by `pip`, the `yotta login` command results in AttributeError.

```
$ yotta
usage: yotta [-h] [--version] [-v] [--plain] [--colourful] [--noninteractive]
             [-t TARGET] [--config path/to/config.json]
             <subcommand> ...
yotta: error: too few arguments
$ yotta login
From cffi callback <function _verify_callback at 0x7fa122d38758>:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/OpenSSL/SSL.py", line 313, in wrapper
    _lib.X509_up_ref(x509)
AttributeError: 'module' object has no attribute 'X509_up_ref'

You need to log in to do this.
Press enter to continue.
Your browser will open to complete login.
^Cwarning: interrupted
$ 
```

sgk, 2017-12-03:
I found this can be corrected by installing latest pyopenssl separately.

```
$ pip install pyopenssl yotta
```

sgk, 2017-12-03:
You may want to add note to the installation page http://docs.yottabuild.org/#installing-on-linux

thegecko, 2017-12-19:
Similar issues have also been fixed using:

    $ pip install --upgrade setuptools
    $ pip install --upgrade pyOpenSSL

PR #826 (2018-01-22): Update cryptography requirement
This PR replaces #824 and updates the cryptography requirement for yotta. I believe it also fixes #823.

Kudos to @cmonr for the `setuptools` upgrade idea over at #825
closed 2018-01-22 by thegecko