ARMmbed/yotta #835: pip install yotta fails with pip v10 (due to project_generator dependency)

microbit-carlos, 2018-04-18:
Please feel free to close this issue, as the problem is not directly caused by yotta itself, but since it will affect users trying to install yotta I though it would be useful to have some info and visibility here.

pip install with pip v10 (released on the 14th of April) will fail when trying to install dependency `project_generator`:

```
Collecting project_generator<0.9.0,>=0.8.0 (from valinor<1.0,>=0.0.0->yotta)
  Downloading https://files.pythonhosted.org/packages/b4/bb/91ad20d17485d01829453767323dd2d13b325a0c7154bf5f8f375cbaac1e/project_generator-0.8.16.zip (86kB)
    100% |████████████████████████████████| 92kB 7.1MB/s 
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/private/var/folders/l1/szsxrhfj7v1fglxfv9b35lz80000gn/T/pip-install-wJUzJQ/project-generator/setup.py", line 18, in <module>
        from pip.req import parse_requirements
    ImportError: No module named req
```

This has already been reported in the `project_generator` issue tracker, with more info over there: https://github.com/project-generator/project_generator/issues/440

0xc0170, 2018-04-19:
As progen was updated (thanks @microbit-carlos for the fix).

Valinor should probably be updated to use the latest version (there's now 'project_generator>=0.8.0,<0.9.0',) and the fix is in 0.9.11

I can even propagate the fix to 0.8.x ?

0xc0170, 2018-06-11:
I believe this can be closed?

carlosperate, 2018-06-11:
If the dependencies have been fixed and their version updated in master it hasn't been published to PyPI yet. On a new environment with pip 10.0.1:

```
...
Collecting project_generator<0.9.0,>=0.8.0 (from valinor<1.0,>=0.0.0->yotta)
  Cache entry deserialization failed, entry ignored
  Downloading https://files.pythonhosted.org/packages/b4/bb/91ad20d17485d01829453767323dd2d13b325a0c7154bf5f8f375cbaac1e/project_generator-0.8.16.zip (86kB)
    100% |████████████████████████████████| 92kB 3.1MB/s 
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/private/var/folders/l1/szsxrhfj7v1fglxfv9b35lz80000gn/T/pip-install-d82c8koo/project-generator/setup.py", line 18, in <module>
        from pip.req import parse_requirements
    ModuleNotFoundError: No module named 'pip.req'
```

Probably better to keep it open until `pip install yotta` works.

ccrov, 2018-07-13:
when i use "pip install yotta " : what i should do ?
Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/private/var/folders/rc/78dl9vfd3fd0ppwjl0zjlph00000gn/T/pip-install-gq_hc0/project-generator/setup.py", line 18, in <module>
        from pip.req import parse_requirements
    ImportError: No module named req
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /private/var/folders/rc/78dl9vfd3fd0ppwjl0zjlph00000gn/T/pip-install-gq_hc0/project-generator/
allans-MacBook-Air:pxt allan$

microbit-carlos, 2018-07-13:
The easiest way to get around it at the moment is to install an older version of pip, the latest v9 should work.

0xc0170, 2018-07-13:
I published 0.8.17 that addresses this issue (cherry picked fix from 0.9.x)

microbit-carlos, 2018-07-13:
Just tried it and `pip install yotta` now works again, thanks @0xc0170!

closed 2018-07-13 by microbit-carlos