ARMmbed/yotta #839: python setup.py egg_info" failed with error

ccrov, 2018-07-13:
when i use pip install yotta；but it can not install 
### 
Collecting project_generator<0.9.0,>=0.8.0 (from valinor<1.0,>=0.0.0->yotta)
  Using cached https://files.pythonhosted.org/packages/b4/bb/91ad20d17485d01829453767323dd2d13b325a0c7154bf5f8f375cbaac1e/project_generator-0.8.16.zip
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/private/var/folders/rc/78dl9vfd3fd0ppwjl0zjlph00000gn/T/pip-install-gq_hc0/project-generator/setup.py", line 18, in <module>
        from pip.req import parse_requirements
    ImportError: No module named req
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /private/var/folders/rc/78dl9vfd3fd0ppwjl0zjlph00000gn/T/pip-install-gq_hc0/project-generator/
allans-MacBook-Air:pxt allan$

0xc0170, 2018-07-13:
This was reported earlier. Pip 10 update broke some modules, this is the case. Please install project generator 0.9.11.  That will fix the issue.

0xc0170, 2018-07-13:
By default, via valinor, you get version 0.8.x. We might want to update that to resolve this

0xc0170, 2018-07-13:
Duplicate #835, please refer to the original one. I published the fix

This can be closed

ccrov, 2018-07-13:
ok thank you