ARMmbed/yotta #840: Select build tool

jamesadevine, 2018-09-19:
Hiya,

Up until recently yotta was using `make` as the build tool on mac which generates colourful compiler errors that are far more visible than ninja.

Is there a way to set the build tool for yotta? I would like to revert to using `make` as my default.

jamesadevine, 2018-09-19:
@thegecko When I say recent, I mean within the last four months.

It was when I upgrade to `0.18.5`.

thegecko, 2018-09-19:
I don't believe there have been any changes to yotta for at least 6 months :)

I can't see where a change like this was made. Yotta uses cmake with ninja files and has done for years AFAIK

What tool are you seeing in use?

jamesadevine, 2018-09-19:
@thegecko It uses cmake + ninja. When Yotta first came out it used native mac make + ninja, so I guess when I upgraded I got the latest version with cmake + ninja (I hadn't upgraded in a while...). 

I would assume that given that it has historically been cmake + ninja, there would be no way to add the ability to set it to another build tool? 

It's just a simple change to the cmake invocation command:
```py
system("cmake .. -DCMAKE_BUILD_TYPE=RelWithDebInfo -G \"<INSERT BUILD TOOL HERE>\"")
```

jamesadevine, 2018-09-19:
It looks like some sort of argument passing is already supported:

https://github.com/ARMmbed/yotta/blob/986cfab25040f02b142f00823417fb374fd23bfc/yotta/lib/target.py#L494

thegecko, 2018-09-19:
I'm not aware of an official way to switch it, however arguments can be passed and may do what you need.
If not, PRs are always welcome!

jamesadevine, 2018-09-19:
Happy to do a PR, but would like to learn of the best approach before hand. It would make sense for it to be a static configuration too – is that supported?

jamesadevine, 2018-09-20:
@thegecko This is actually already supported:

`yt build -G "Unix Makefiles"`

Closing!

closed 2018-09-20 by jamesadevine

thegecko, 2018-09-20:
Awesome, glad you found that :)