ARMmbed/yotta #841: Doesn't find tag in yotta install

notexactlyawe, 2018-09-23:
I'm trying to install the `v2.0.0-rc10` tag of `lancaster-university\microbit` but I'm running into the following error:

```
$ yotta install lancaster-university/microbit#v2.0.0-rc10
info: microbit, lancaster-university/microbit#v2.0.0-rc10
info: get versions for microbit
error: Github repository "lancaster-university/microbit#v2.0.0-rc10" does not have any tags, branches or commits matching "v2.0.0-rc10"
```

I can confirm this tag exists, because it has a page on GitHub [here](https://github.com/lancaster-university/microbit-dal/releases/tag/v2.0.0-rc10).

My theory is that it doesn't like the `-` in the URL, because when I then tried to install `v2.1.0` it ran successfully:

```
$ yotta install lancaster-university/microbit#v2.1.0
info: microbit, lancaster-university/microbit#v2.1.0
info: get versions for microbit
info: download microbit@v2.1.0 from GitHub lancaster-university/microbit
info: dependency microbit: lancaster-university/microbit#v2.1.0 written to module.json
info: get versions for microbit-dal
info: download microbit-dal@v2.1.0 from GitHub lancaster-university/microbit-dal
info: get versions for mbed-classic
info: download mbed-classic@microbit_hfclk+mb6 from GitHub lancaster-university/mbed-classic
info: get versions for ble
info: download ble@v2.5.0+mb3 from GitHub lancaster-university/BLE_API
info: get versions for ble-nrf51822
info: download ble-nrf51822@v2.5.0+mb7 from GitHub lancaster-university/nRF51822
info: get versions for nrf51-sdk
info: download nrf51-sdk@v2.2.0+mb4 from GitHub lancaster-university/nrf51-sdk
```

thegecko, 2018-09-23:
Maybe use microbit-dal?
```
$ yotta install lancaster-university/microbit-dal#v2.0.0-rc10
```

notexactlyawe, 2018-09-24:
Ah, looks like I mistook the two libraries, having now run on the `microbit-dal` repo as you suggested I don't think there is an issue with yotta. Thanks for your speedy response!

closed 2018-09-24 by notexactlyawe