ARMmbed/yotta #842: cryptography 2.1.4 which is incompatible.

jasonlamkk, 2018-11-15:
The minimum cryptography version is now 2.2.1.
according to https://pypi.org/project/pyOpenSSL/

which blocks pip install yotta

```
in setup.py

        'cryptography>=2.1.4,<2.2',
```
can we upgrade to newer cryptography, not <2.2. ?

not sure if already have incompatible code

thegecko, 2018-11-15:
PR welcome :)

jasonlamkk, 2018-11-17:
found that it can be installed with following command: 
`pip install yotta --upgrade cryptography`

this issue can be closed, may be great if update the Mac section of http://docs.yottabuild.org/

pandasauce, 2019-03-13:
This also affects Linux; Ubuntu 18.04 in my case.

Out of the box:

```
pyopenssl 19.0.0 has requirement cryptography>=2.3, but you'll have cryptography 2.1.4 which is incompatible.
pyocd 0.17.0 has requirement pyyaml<5.0,>=4.2b1, but you'll have pyyaml 3.13 which is incompatible.
```

With `--upgrade cryptography`:

```
yotta 0.18.5 has requirement cryptography<2.2,>=2.1.4, but you'll have cryptography 2.6.1 which is incompatible.
```

0xc0170, 2019-06-04:
Wasnt this fixed by #847 ?

jasonlamkk, 2019-06-04:
May be, the fix was posted after this issue.

thegecko, 2019-06-04:
I believe so, please reopen this issue if not :)

closed 2019-06-04 by thegecko