ARMmbed/yotta #846: Broken dependencies

pandasauce, 2019-03-13:
```
$ pip install yotta
pyocd 0.17.0 has requirement pyyaml<5.0,>=4.2b1, but you'll have pyyaml 3.13 which is incompatible.
```

Looks like some other dependency pulls in PyOCD, which requires a version of PyYAML that causes conflicts with even more packages:

```
$ pip install PyYAML==4.2b1  
Collecting PyYAML==4.2b1
pyocd 0.17.0 requires future, which is not installed.
pyocd 0.17.0 requires intervaltree<4.0,>=3.0.2, which is not installed.
pyocd 0.17.0 requires pyusb<2.0,>=1.0.0b2, which is not installed.
pyocd 0.17.0 requires websocket-client, which is not installed.
valinor 0.0.15 has requirement pyyaml<4,>=3, but you'll have pyyaml 4.2b1 which is incompatible.
```

pandasauce, 2019-03-13:
I raised a [PR on Valinor](https://github.com/ARMmbed/valinor/pull/31) to resolve this, however because their last version bump went over the `<1.0` [boundary set by yotta](https://github.com/ARMmbed/yotta/blob/master/setup.py#L68), it might require another PR for yotta to pull in the new version of Valinor with the fix.

0xc0170, 2019-05-03:
@thegecko can you bump valinor to 1.1.x req in this repo ?

0xc0170, 2019-05-03:
@pandasauce thanks for reporting this 👍

pandasauce, 2019-05-04:
I did a quick test with the fixes and it looks like since this was opened PyOCD raised their PyYAML dependency again, which requires Valinor and #847 to be updated again to avoid dependency conflicts.

```
error: PyYAML 4.2b4 is installed but pyyaml<6.0,>=5.1 is required by set(['pyOCD'])
```

pandasauce, 2019-05-04:
Made the PR to Valinor: https://github.com/ARMmbed/valinor/pull/33

0xc0170, 2019-05-06:
Valinor 1.1.4 released

PR #847 (2019-05-09): Fix broken dependencies
This is meant to resolve #846 but is not ready to merge until https://github.com/ARMmbed/valinor/pull/31 gets merged
closed 2019-05-09 by thegecko