ARMmbed/yotta #856: `yotta target` fails with `error: 'RegistryThingVersion' object has no attribute 'truncate'`

WillSewell, 2019-09-01:
macOS Mojave 10.14.06
Python 3.7.4
yotta 0.19.0
cmake 3.15.2
ninja 1.9.0
arm-none-eabi-gcc stable 8-2018-q4-major

I'm trying to build https://github.com/lancaster-university/microbit-samples. When I run

```
yotta -vvv target bbc-microbit-classic-gcc
```

I get the output

```
debug:root: setProperty: build.target <class 'str'>:bbc-microbit-classic-gcc,*
debug:root: write settings to "./.yotta.json" (will ensure directory "." exists)
debug:root: setProperty: build.targetSetExplicitly <class 'bool'>:True
debug:root: write settings to "./.yotta.json" (will ensure directory "." exists)
debug:target: satisfy target: bbc-microbit-classic-gcc,*
debug:access: check path /Users/will/src/microbit-samples/yotta_targets/bbc-microbit-classic-gcc for bbc-microbit-classic-gcc
debug:access: got INVALID COMPONENT @ /Users/will/src/microbit-samples/yotta_targets/bbc-microbit-classic-gcc: OrderedDict()
debug:access: not found bbc-microbit-classic-gcc locally
info: get versions for bbc-microbit-classic-gcc
debug:access: satisfy bbc-microbit-classic-gcc from targets registry
debug:access: fingerprint: 30:ad:0f:ad:f8:42:8b:73:b2:47:ac:69:cf:07:4b:91
debug:access: token fields: {'iss': 'yotta', 'aud': 'https://registry.yottabuild.org', 'prn': '30:ad:0f:ad:f8:42:8b:73:b2:47:ac:69:cf:07:4b:91', 'exp': 1567344868}
debug:access: encoded token: eyJ0eXAiOiJKV1QiLCJhbGciOiJSUzI1NiJ9.eyJpc3MiOiJ5b3R0YSIsImF1ZCI6Imh0dHBzOi8vcmVnaXN0cnkueW90dGFidWlsZC5vcmciLCJwcm4iOiIzMDphZDowZjphZDpmODo0Mjo4Yjo3MzpiMjo0NzphYzo2OTpjZjowNzo0Yjo5MSIsImV4cCI6MTU2NzM0NDg2OH0.dzpznHPoYHcMIkY9vXtHdWaAbqZWJl_ZvkQLc4KEHEbi2RXB6-Wm7NNrmh7spDJbrT4YyWmmZavG90-jD1JN_Z9XmfPI6Wyw_DeBKX-3shMu0AUBWnYzcGrwSDogQmbl5emkhaN8IisCtHdYNRAQsBPSk98VJKKlmjODGuc_OzEftI3Kgl7WN5SD4Mmt8HmrwszpvhGOCC7Y4CPOs4WEyLpr-bQ2RZeNk0lSkANJ-iVC1rqcQ28D7rOwhsvyrC51GvDKLrRLbWpma47cGCIsDeuUwICeDI-6XfEZQfhWLqS0tP3TvcCf1oLOW8VYesNYHkqEk6t_QKam3E0ZSQbNCQ
debug:access: GET https://registry.yottabuild.org/targets/bbc-microbit-classic-gcc/versions, {'Authorization': 'Bearer eyJ0eXAiOiJKV1QiLCJhbGciOiJSUzI1NiJ9.eyJpc3MiOiJ5b3R0YSIsImF1ZCI6Imh0dHBzOi8vcmVnaXN0cnkueW90dGFidWlsZC5vcmciLCJwcm4iOiIzMDphZDowZjphZDpmODo0Mjo4Yjo3MzpiMjo0NzphYzo2OTpjZjowNzo0Yjo5MSIsImV4cCI6MTU2NzM0NDg2OH0.dzpznHPoYHcMIkY9vXtHdWaAbqZWJl_ZvkQLc4KEHEbi2RXB6-Wm7NNrmh7spDJbrT4YyWmmZavG90-jD1JN_Z9XmfPI6Wyw_DeBKX-3shMu0AUBWnYzcGrwSDogQmbl5emkhaN8IisCtHdYNRAQsBPSk98VJKKlmjODGuc_OzEftI3Kgl7WN5SD4Mmt8HmrwszpvhGOCC7Y4CPOs4WEyLpr-bQ2RZeNk0lSkANJ-iVC1rqcQ28D7rOwhsvyrC51GvDKLrRLbWpma47cGCIsDeuUwICeDI-6XfEZQfhWLqS0tP3TvcCf1oLOW8VYesNYHkqEk6t_QKam3E0ZSQbNCQ', 'X-Yotta-Client-Version': '0.19.0', 'X-Yotta-Client-ID': 'e869a51f-3a62-42ab-a674-f8e9baf7ce3e'}
debug:urllib3.connectionpool: Starting new HTTPS connection (1): registry.yottabuild.org:443
debug:urllib3.connectionpool: https://registry.yottabuild.org:443 "GET /targets/bbc-microbit-classic-gcc/versions HTTP/1.1" 200 7031
debug:access: RegistryThingVersion targets/bbc-microbit-classic-gcc data: OrderedDict([('scripts', OrderedDict([('test', ['mbed_test_wrapper', '--target', 'NRF51822', '$program']), ('debug', ['valinor', '--target', 'NRF51822', '$program'])])), ('toolchain', 'CMake/toolchain.cmake'), ('similarTo', ['nrf51822', 'nrf', 'nordic', 'cortex-m0', 'armv6-m', 'mbed-classic', 'microbit']), ('keywords', ['mbed-target:nrf51822', 'mbed-official', 'nrf51822', 'nordic', 'gcc']), ('inherits', OrderedDict([('mbed-gcc', '0.1.3')])), ('licenses', [OrderedDict([('type', 'Apache-2.0'), ('url', 'https://spdx.org/licenses/Apache-2.0')])]), ('description', 'Official mbed build target for the BBC micro:bit, using the gcc toolchain.'), ('version', '0.2.3'), ('name', 'bbc-microbit-classic-gcc'), ('readme_url', 'https://yotta.blob.core.windows.net/targets/e5c64894601062df2b3de4cecde103ad472c218b-bbc-microbit-classic-gcc-0.2.3-readme.md'), ('hash', OrderedDict([('sha256', 'b789a3edd1077cf8af55e804aec16b4cfcbc9dd8e0406231ac1a259d15f206d1')])), ('created', '2016-08-05T10:27:14.978Z'), ('type', 'target')])
debug:access: RegistryThingVersion targets/bbc-microbit-classic-gcc data: OrderedDict([('scripts', OrderedDict([('test', ['mbed_test_wrapper', '--target', 'NRF51822', '$program']), ('debug', ['valinor', '--target', 'NRF51822', '$program'])])), ('toolchain', 'CMake/toolchain.cmake'), ('similarTo', ['nrf51822', 'nrf', 'nordic', 'cortex-m0', 'armv6-m', 'mbed-classic', 'microbit']), ('keywords', ['mbed-target:nrf51822', 'mbed-official', 'nrf51822', 'nordic', 'gcc']), ('inherits', OrderedDict([('mbed-gcc', '0.1.3')])), ('licenses', [OrderedDict([('type', 'Apache-2.0'), ('url', 'https://spdx.org/licenses/Apache-2.0')])]), ('description', 'Official mbed build target for the BBC micro:bit, using the gcc toolchain.'), ('version', '0.2.1'), ('name', 'bbc-microbit-classic-gcc'), ('readme_url', 'https://yotta.blob.core.windows.net/targets/4e50fc2eb58972d973b73e3fb435e10d665e0d9a-bbc-microbit-classic-gcc-0.2.1-readme.md'), ('hash', OrderedDict([('sha256', '19645fd7e1eb71c1f88631944e9d4f812f8ac403f5b721399b909855d3edb191')])), ('created', '2016-08-05T10:26:38.230Z'), ('type', 'target')])
debug:access: RegistryThingVersion targets/bbc-microbit-classic-gcc data: OrderedDict([('scripts', OrderedDict([('test', ['mbed_test_wrapper', '--target', 'NRF51822', '$program']), ('debug', ['valinor', '--target', 'NRF51822', '$program'])])), ('toolchain', 'CMake/toolchain.cmake'), ('similarTo', ['nrf51822', 'nrf', 'nordic', 'cortex-m0', 'armv6-m', 'mbed-classic', 'microbit']), ('keywords', ['mbed-target:nrf51822', 'mbed-official', 'nrf51822', 'nordic', 'gcc']), ('inherits', OrderedDict([('mbed-gcc', '0.1.3')])), ('licenses', [OrderedDict([('type', 'Apache-2.0'), ('url', 'https://spdx.org/licenses/Apache-2.0')])]), ('description', 'Official mbed build target for the mbed nrf51822 development board, using the armgcc toolchain.'), ('version', '0.2.0'), ('name', 'bbc-microbit-classic-gcc'), ('readme_url', 'https://yotta.blob.core.windows.net/targets/27a65009b5b7f4e845e05ab4fdefe8623b777405-bbc-microbit-classic-gcc-0.2.0-readme.md'), ('hash', OrderedDict([('sha256', '73a87511ad66c05590d5912758f72d686737abc391579c6daa52737c3549d459')])), ('created', '2016-01-05T14:39:05.912Z'), ('type', 'target')])
debug:access: RegistryThingVersion targets/bbc-microbit-classic-gcc data: OrderedDict([('scripts', OrderedDict([('test', ['mbed_test_wrapper', '--target', 'NRF51822', '$program']), ('debug', ['valinor', '--target', 'NRF51822', '$program'])])), ('toolchain', 'CMake/toolchain.cmake'), ('similarTo', ['nrf51822', 'nrf', 'nordic', 'cortex-m0', 'armv6-m', 'mbed-classic', 'microbit']), ('keywords', ['mbed-target:nrf51822', 'mbed-official', 'nrf51822', 'nordic', 'gcc']), ('inherits', OrderedDict([('mbed-gcc', '0.1.3')])), ('licenses', [OrderedDict([('type', 'Apache-2.0'), ('url', 'https://spdx.org/licenses/Apache-2.0')])]), ('description', 'Official mbed build target for the mbed nrf51822 development board, using the armgcc toolchain.'), ('version', '0.1.4'), ('name', 'bbc-microbit-classic-gcc'), ('readme_url', 'https://yotta.blob.core.windows.net/targets/22c0b6b2340c30550c048f4aedad79d87c6c5da4-bbc-microbit-classic-gcc-0.1.4-readme.md'), ('hash', OrderedDict([('sha256', 'de92d828e154c7af241acc681015e2c4dfd89ac179bb1d74f04f09345cbac23f')])), ('created', '2016-01-01T14:25:58.350Z'), ('type', 'target')])
debug:access: RegistryThingVersion targets/bbc-microbit-classic-gcc data: OrderedDict([('scripts', OrderedDict([('test', ['mbed_test_wrapper', '--target', 'NRF51822', '$program']), ('debug', ['valinor', '--target', 'NRF51822', '$program'])])), ('toolchain', 'CMake/toolchain.cmake'), ('similarTo', ['nrf51822', 'nrf', 'nordic', 'cortex-m0', 'armv6-m', 'mbed-classic', 'microbit']), ('keywords', ['mbed-target:nrf51822', 'mbed-official', 'nrf51822', 'nordic', 'gcc']), ('inherits', OrderedDict([('mbed-gcc', '*')])), ('licenses', [OrderedDict([('type', 'Apache-2.0'), ('url', 'https://spdx.org/licenses/Apache-2.0')])]), ('description', 'Official mbed build target for the mbed nrf51822 development board, using the armgcc toolchain.'), ('version', '0.1.3'), ('name', 'bbc-microbit-classic-gcc'), ('readme_url', 'https://yotta.blob.core.windows.net/targets/0c87a3b4c2a83625ec8408c6cb9dd10e6db34a9e-bbc-microbit-classic-gcc-0.1.3-readme.md'), ('hash', OrderedDict([('sha256', 'b5a903873f12c54988a8530ac98100194b5e90a874949c42da556991bd982862')])), ('created', '2015-08-19T17:41:52.697Z'), ('type', 'target')])
[... cut ...] he armgcc toolchain.'), ('version', '0.1.0'), ('name', 'bbc-microbit-classic-gcc'), ('readme_url', 'https://yotta.blob.core.windows.net/targets/16890edd581265d5133b305dc2d9bbc546e85acd-bbc-microbit-classic-gcc-0.1.0-readme.md'), ('hash', OrderedDict([('sha256', 'cbf4cbdae3a8c0b17b4bd1e7b16e7b0b6126b4019b792bfed6911b170ecb7832')])), ('created', '2015-07-30T01:43:20.077Z'), ('type', 'target')])
error: 'RegistryThingVersion' object has no attribute 'truncate'
```

Any ideas wha is causing this?
```

thegecko, 2019-09-02:
It's likely the python version is too new

WillSewell, 2019-09-02:
Thanks for the suggestion. I tried 3.6.9, 3.5.7. and 3.4.10 but unfortunately I get the same error.

Mac898, 2019-09-05:
Hi, I am experiencing the exact same issue. Just wondering if there are any more details I can provide to try and help with this. I have tried with python 3.8, 3.7, 3.6, 3.5 and 3.3 as well as 2.7. Any updates would be much appreciated. 

My log is attached:

[log-github.txt](https://github.com/ARMmbed/yotta/files/3578639/log-github.txt)

thegecko, 2019-09-05:
Any idea when this issue started to occur?

Is it something recent?

WillSewell, 2019-09-05:
I'm afraid I don't know. It's my first time using yotta.

psch2, 2019-09-05:
Hi,
The cause of the issue is a call to _truncate_ in _semantic_version/base.py_. The variable _version_ points to an instance of class _yotta.lib.registry_access.RegistryThingVersion_ which has no _truncate_ method defined:
```
> /home/build/.local/lib/python3.7/site-packages/semantic_version/base.py(894)match()
-> version = version.truncate('prerelease')
(Pdb) l
889  	        self.build_policy = self.BUILD_STRICT if target.build else build_policy
890  	
891  	    def match(self, version):
892  	        if self.build_policy != self.BUILD_STRICT:
893  	            import pdb; pdb.set_trace()
894  ->	            version = version.truncate('prerelease')
895  	
896  	        if version.prerelease:
897  	            same_patch = self.target.truncate() == version.truncate()
898  	
899  	            if self.prerelease_policy == self.PRERELEASE_SAMEPATCH and not same_patch:
(Pdb) interact
*interactive*
>>> version
bbc-microbit-classic-gcc@0.2.3 from the public module registry
>>> type(version)
<class 'yotta.lib.registry_access.RegistryThingVersion'>
>>> 
now exiting InteractiveConsole...
(Pdb) n
AttributeError: 'RegistryThingVersion' object has no attribute 'truncate'
```

psch2, 2019-09-05:
Got it working by adding the following method to class _Version_ defined in _yotta/lib/version.py_:
```
    def truncate(self, level):
        return self.version.truncate(level)

```

thegecko, 2019-09-06:
@psch2 great work, please open a PR and we can check all tests are still good.

lapointexavier, 2019-11-16:
@thegecko Would it be possible to get a new release to include 047cff2? I am a new user, just installed 0.19.0, and I'm affected by that bug. Thanks!

thegecko, 2019-11-16:
I've released a newer version to TestPyPI:

https://test.pypi.org/project/yotta/

Could you try it out and confirm it fixes this issue?

lapointexavier, 2019-11-17:
@thegecko It did, thanks!

thegecko, 2019-11-17:
OK, thanks.

I've released v0.20.0

closed 2019-11-17 by thegecko