ARMmbed/yotta #860: pip install -U yotta : errors out in Win10 [duplicate]

monojk, 2020-02-20:
When trying to install yotta on Win10 I get the errors at the end. Probable cause:   
**It requires requests[security] and cryptography>=2.1.4,<=2.3. At the same time requests requires pyOpenSSL which requires cryptography>=2.8**

From the installation:
`Collecting yotta
  Using cached yotta-0.20.0.tar.gz (130 kB)
...
Collecting cryptography<=2.3,>=2.1.4
  Using cached cryptography-2.3.tar.gz (449 kB)
...
Collecting pyOpenSSL>=0.14; extra == "security"
  Using cached pyOpenSSL-19.1.0-py2.py3-none-any.whl (53 kB)
...
ERROR: pyopenssl 19.1.0 has requirement cryptography>=2.8, but you'll have cryptography 2.3 which is incompatible.
Installing collected packages: cryptography, pyOpenSSL, requests, PyJWT, wrapt, deprecated, PyGithub, colorama, hgapi, MarkupSafe, Jinja2, pathlib, jsonschema, argcomplete, mbed-test-wrapper, pyyaml, pyelftools, milksnake, appdirs, cmsis-pack-manager, prettytable, sortedcontainers, intervaltree, intelhex, pywinusb, pyusb, pyOCD, argparse, xmltodict, project-generator-definitions, project-generator, valinor, jsonpointer, ntfsutils, yotta
    Running setup.py install for cryptography ... error
...
    build\temp.win32-3.8\Release\_openssl.c(546): fatal error C1083: Cannot open include file: 'openssl/opensslv.h': No such file or directory
    error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\BIN\\cl.exe' failed with exit status 2`

thegecko, 2020-02-20:
PRs Welcome :)

0xc0170, 2020-02-28:
I was able to reproduce the problem :/ @monojk Have you already found a fix - bumped the version and tested it?

monojk, 2020-02-28:
No, I don't have a fix. I'm just a user or trying to use Yotta for compiling micropython.

monojk, 2020-03-06:
Any progress on this?

thegecko, 2020-03-06:
@rwalton-arm Does your fix close this issue?

rwalton-arm, 2020-03-06:
Yeah, my fix should have resolved this. I tested Windows 10 python 3.7 and 2.7 and the installation completed without issue.

rwalton-arm, 2020-03-09:
Fix is in v0.20.5. Please let us know if you have any more issues.

closed 2020-03-09 by rwalton-arm

monojk, 2020-03-10:
Successfully installed. Thanks